MrChromebox/firmware #418: possible Lillipup regression

bcb-git, 2022-12-30:
Hello there - I think I have found a regression between the UEFI firmware and the RW_LEGACY firmware for the LILLIPUP boards. The audio device is detected in dmesg and lspci on RW_LEGACY (and works fine with existing kernel drivers and the sof_firmware package) but is not found at all on the UEFI firmware. I have the dmesg and lspci (and lspci -n) for the RW_LEGACY firmware and will grab the same from the UEFI firmware for comparison.
[dmesg.rw.txt](https://github.com/MrChromebox/firmware/files/10323684/dmesg.rw.txt)
[lspci.rw.txt](https://github.com/MrChromebox/firmware/files/10323685/lspci.rw.txt)
[lspci-n.rw.txt](https://github.com/MrChromebox/firmware/files/10323686/lspci-n.rw.txt)

bcb-git, 2022-12-30:
Here are the same from the UEFI firmware on the same device. There are two PCI devices which exist on the RW_LEGACY firmware which do not show up on the UEFI firmware:
00:0a.0 1180: 8086:9a0d - "Signal processing controller: Intel Corporation Tigerlake Telemetry Aggregator Driver (rev 01)"
00:1f.3 0401: 8086:a0c8 - "Multimedia audio controller: Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller (rev 20)"

[dmesg.uefi.txt](https://github.com/MrChromebox/firmware/files/10323725/dmesg.uefi.txt)
[lspci.uefi.txt](https://github.com/MrChromebox/firmware/files/10323726/lspci.uefi.txt)
[lspci-n.uefi.txt](https://github.com/MrChromebox/firmware/files/10323728/lspci-n.uefi.txt)

MrChromebox, 2022-12-30:
LILLIPUP and other TGL audio has already been fixed in test builds, update will be released soon

bcb-git, 2022-12-30:
Neat! Thanks =)

walmartshopper, 2022-12-31:
Looking forward to the audio fix.  I have been running the 4.16 release on LILLIPUP-MQUZ for a while in RW_LEGACY. I started having suspend issues and was about to try 4.18.2 full rom but saw another issue here about audio not working. Will test once the update is released.

MrChromebox, 2023-02-03:
fixed in 4.19 update

closed 2023-02-03 by MrChromebox

marcosilva85, 2023-02-11:
invalid closing reason - issue still active. (Issue could be reopened) or just follow-up in new issue: #430