MrChromebox/firmware #419: Digitizer/stylus not working on Lenovo 500e 2nd gen (81MC)

dewitte-77, 2023-01-01:
These devices have a garaged stylus that works in chromeos and is most likely based on the Wacom technology (could not confirm that).
Linux is working  with multiple distro's tested (arch, ubuntu 22.10, opensuse 15.4) and supporting almost all the hardware except for this stylus.
The touch screen itself is working when using fingers but when using the stylus there is no response at all.

I also cannot find any "unknown" device that might represent the digitizer.
Is it possible that this device is not activated through the firmware?

What information could help to diagnose?

mpeter50, 2023-01-06:
I have experienced this with the same model, but haven't got around yet to figure out whats wrong.
I'll follow this issue and try to help if I can.

ellyq, 2023-01-08:
As mentioned, octopus (GLK baseboard) is missing pen power sequencing.
I tried to fix it and gave one user test rom few months back, but it didn't work. They were worried about bricking their machine, as they didn't have SuzyQ, so they gave up on that functionality.

Neither MrChromebox, CoolStar or me have any GLK-based machines with pen support to test, so someone with such machine would need to build SuzyQ..
Please attach logs from the script, as well as cbmem. Make sure to remove any private information (such as WiFi AP MAC in dmesg).

```
#!/bin/bash

printf 'Welcome.\nThis script will ask you for sudo password, which is necessary to extract required information.\n'

# Get current username before escalating permissions
user=$USER

# Create directory to store logs
mkdir -p ~/Desktop/debug-logs
cd ~/Desktop/debug-logs

# Download cbmem and mark it as executable
wget https://elly.rocks/tmp/coreboot-development/cbmem
chmod +x cbmem

# Priviledge escalation [!!!]
sudo su <<EOF

# Grab logs and redirect output to files instead of stdout
dmesg >> dmesg.log

if [ ! -f /sbin/lspci ]
then
	printf 'lspci not found. Please install pciutils.'
else
	lspci -vvvnn >> lspci.log
fi

if [ ! -f /bin/lsusb ]
then 
	printf 'lsusb not found. Please install usbutils.'
else
	lsusb -v >> lsusb.log
fi

if [ ! -f /sbin/dmidecode ]
then
	printf 'Dmidecode not found. Please install it'.
else
	dmidecode > dmidecode.log
fi

## Copy ACPI tables
cp /sys/firmware/acpi/tables/DSDT .
cp /sys/firmware/acpi/tables/SSDT* .

# Grab coreboot logs
./cbmem -c > cbmem.log

# Set file permissions for regular user
chown $user:$user *
chmod 755 *

EOF

# Pack logs into archive and remove temporary folder that stores them
cd ..
tar -czf debug-logs.tar.gz debug-logs
rm -r debug-logs

printf 'Log collection done.\nPlease upload "debug-logs.tar.gz" for analysis.\n'
```

mpeter50, 2023-01-08:
> Neither MrChromebox, CoolStar or me have any GLK-based machines with pen support to test, so someone with such machine would need to build SuzyQ.

I don't have such a cable either, but wanted to have one anyways, so I could build one and help with this.
However it might take some time to do so, as last time I tried I had problems with sourcing parts, so if dewitte-77 has one already that would probably be faster. I'll now prioritize this, though.

dewitte-77, 2023-01-08:
I have ordered the necessary parts to build the cable.
Just for my understanding: do we only need the cable when things go wrong? (for example, the test firmware does not work and we have to use the cable to unbrick)
So I can already execute the script and upload the requested file for you guys to create a test firmware?

ellyq, 2023-01-09:
Yes please, it will provide crucial information for troubleshooting.

MrChromebox, 2023-01-09:
one does not need a Suzy-Q cable to debug this. If the device isn't showing up in cbmem under the SSDT, then the ACPI code isn't present for the OS driver to attach

dewitte-77, 2023-01-09:
[debug-logs.tar.gz](https://github.com/MrChromebox/firmware/files/10376646/debug-logs.tar.gz)
Had to build cbmem because the download link gave errors 
Hope this contains the information you need.

MrChromebox, 2023-01-09:
looking at the code for the 4.18.x release, it seems that the reset line for the digitizer/pen was not being released prior to detection, so the ACPI wasn't getting added. This was corrected when upstreaming, and should be fixed in the upcoming 4.19 release. If you ping me on the chrultrabook Discord, I can whip up a test build for you to try

closed 2023-01-10 by dewitte-77