MrChromebox/firmware #421: Pixel Slate (Nocturne) coreboot crash

minego, 2023-01-12:
`[  122.266594] ------------[ cut here ]------------
[  122.266596] memcpy: detected field-spanning write (size 248) of single field "&device->entry" at drivers/firmware/google/coreboot_table.c:104 (size 8)
[  122.266615] WARNING: CPU: 3 PID: 262 at drivers/firmware/google/coreboot_table.c:104 coreboot_table_probe+0x1ba/0x1e2 [coreboot_table]
[  122.266623] Modules linked in: coreboot_table(+) mac_hid crypto_user fuse bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 mmc_block i915 drm_buddy intel_gtt drm_display_helper sdhci_pci cqhci cec sdhci video ttm cros_ec_keyb mmc_core crc32c_intel xhci_pci xhci_pci_renesas wmi hid_google_hammer(+) hid_vivaldi_common vivaldi_fmap matrix_keymap
[  122.266655] CPU: 3 PID: 262 Comm: systemd-udevd Tainted: G      D            6.1.4-zen2-1-zen #1 0e2b1da333fe8d418d1a949f82e5d54d9731d1e0
[  122.266658] Hardware name: Google Nocturne/Nocturne, BIOS MrChromebox-4.18.2 11/29/2022
[  122.266660] RIP: 0010:coreboot_table_probe+0x1ba/0x1e2 [coreboot_table]
[  122.266666] Code: 08 00 00 00 48 89 c6 48 89 04 24 48 c7 c2 00 a1 a0 c0 48 c7 c7 48 a1 a0 c0 4c 89 44 24 08 c6 05 f3 1e 00 00 01 e8 39 fb 7a fc <0f> 0b 4c 8b 44 24 08 48 8b 04 24 e9 71 ff ff ff 41 bf ea ff ff ff
[  122.266668] RSP: 0018:ffff9c82007cbbb0 EFLAGS: 00010286
[  122.266671] RAX: 0000000000000000 RBX: ffff8fd787b82800 RCX: 0000000000000027
[  122.266672] RDX: ffff8fd8f6da1668 RSI: 0000000000000001 RDI: ffff8fd8f6da1660
[  122.266674] RBP: ffff9c82000ed018 R08: 80000000fffff2df R09: 0000000000000000
[  122.266676] R10: ffffffffffffffff R11: ffffffffbed17e41 R12: 0000000000000000
[  122.266677] R13: ffff9c82000ed000 R14: ffff8fd7811a2010 R15: 0000000000000000
[  122.266678] FS:  00007f240da2b080(0000) GS:ffff8fd8f6d80000(0000) knlGS:0000000000000000
[  122.266680] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  122.266682] CR2: 00007f240d998000 CR3: 0000000100eb6006 CR4: 00000000003706e0
[  122.266684] Call Trace:
[  122.266686]  <TASK>
[  122.266689]  platform_probe+0x45/0x90
[  122.266695]  really_probe+0xdb/0x3f0
[  122.266698]  ? pm_runtime_barrier+0x54/0x90
[  122.266702]  __driver_probe_device+0x78/0x170
[  122.266705]  driver_probe_device+0x1f/0x110
[  122.266708]  __driver_attach+0x98/0x1e0
[  122.266711]  ? __device_attach_driver+0x150/0x150
[  122.266714]  bus_for_each_dev+0x88/0xd0
[  122.266717]  bus_add_driver+0x1b2/0x200
[  122.266726]  driver_register+0x8d/0xe0
[  122.266729]  coreboot_table_driver_init+0x2f/0x1000 [coreboot_table 1a24e21f9952c9f0e16eb9ef1d082202918f4124]
[  122.266736]  ? 0xffffffffc0a0e000
[  122.266742]  do_one_initcall+0x5a/0x220
[  122.266748]  do_init_module+0x4a/0x1e0
[  122.266751]  __x64_sys_init_module+0xc1/0xe0
[  122.266756]  do_syscall_64+0x5c/0x90
[  122.266760]  ? syscall_exit_to_user_mode+0x2c/0x1d0
[  122.266763]  ? do_syscall_64+0x6b/0x90
[  122.266767]  ? syscall_exit_to_user_mode+0x2c/0x1d0
[  122.266770]  ? do_syscall_64+0x6b/0x90
[  122.266772]  ? exc_page_fault+0x74/0x170
[  122.266775]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[  122.266778] RIP: 0033:0x7f240e521eae
[  122.266781] Code: 48 8b 0d dd ee 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d aa ee 0c 00 f7 d8 64 89 01 48
[  122.266783] RSP: 002b:00007ffc7a153eb8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[  122.266785] RAX: ffffffffffffffda RBX: 000055915bc1bf00 RCX: 00007f240e521eae
[  122.266787] RDX: 00007f240e9ef343 RSI: 0000000000003eb6 RDI: 00007f240d978010
[  122.266788] RBP: 00007f240e9ef343 R08: 27d4eb2f165667c5 R09: 85ebca77c2b2ae63
[  122.266790] R10: 00007f240e515aeb R11: 0000000000000246 R12: 0000000000020000
[  122.266791] R13: 000055915bc06fa0 R14: 000055915bc1bf00 R15: 000055915bc1be30
[  122.266794]  </TASK>
[  122.266796] ---[ end trace 0000000000000000 ]---`

MrChromebox, 2023-01-12:
that's not a coreboot crash, it's a kernel driver crash

minego, 2023-01-12:
> that's not a coreboot crash, it's a kernel driver crash

You're right. I'm a bit new to coreboot and easily confused :-)

I'm trying to determine if this is related to any issues I'm running into or if it is a red herring. It looks like this is not happening when I boot the arch linux-lts, so maybe this is a red herring.

closed 2023-01-16 by minego