MrChromebox/firmware #426: Asus CN60 can't enroll in management

getut, 2023-01-31:
I have a CN60 that I am testing to breath life into a bunch more if I can get them to work. I have successfully installed the mrchromebox uefi full rom and then also successfully installed chromeos flex. On first boot I tried to enroll and I got the error message about installation-time attributes lock. I went googling again and found that the TPM had to be cleared, so found instructions and downloaded galliumOS for haswell, booted from that, installed tpm-tools while booted in the live cd, and then ran sudo tpm_clear -f. It said it cleared, so I rebooted when it requested. On the next boot to chromeOS flex, it now said it detected a TPM module and asked me to reboot again. I did, and on next reboot it seemed to boot normally, but I then tried to enroll again and got exactly the same error that I started with. I don't know where to go from here. OH.. it is showing the device in my google management interface but it lists nothing about model and versions.

MrChromebox, 2023-01-31:
I'm honestly not sure what the issue is, other than something related to the TPM, which is not fully functional yet in the UEFI firmware

Straunge, 2023-03-01:
Just writing here to add onto this; I have flashed Flex with full UEFI to a Lenovo N21, N23, and a Dell 11 3180.  All of which work perfectly fine as an individual user.  Fingers crossed we get TPM integration in the UEFI.

MrChromebox, 2023-04-30:
TPM 1.2 admin will be included in the 4.20 release. Tested working across all pre-Skylake Chromebooks

XBGT351, 2023-05-08:
When is 4.20 release scheduled to be released ?

MrChromebox, 2023-05-17:
Fixed in MrChromebox-4.20.0 firmware release

closed 2023-05-17 by MrChromebox