MrChromebox/firmware #428: Acer C733 Bobba chromebook unable to return to stock firmware

thonkinator, 2023-02-03:
I recently installed UEFI/Full ROM firmware, but I'd like to switch back to stock firmware. However, the firmware utility script fails when trying to install it. I'm running vanilla Ubuntu 22.04. Here's the output of firmware-util:
<pre>ChromeOS Device Firmware Utility Script [2023-01-29] 
 (c) Mr Chromebox &lt;mrchromebox@gmail.com&gt; 
<font color="#2AA1B3">*********************************************************</font>
<font color="#2AA1B3">**</font><font color="#A2734C">   Device: </font>Acer Chromebook 311 (BOBBA)
<font color="#2AA1B3">**</font><font color="#A2734C"> Platform: </font>Intel GeminiLake
<font color="#2AA1B3">**</font><font color="#A2734C">  Fw Type: </font>Full ROM / UEFI
<font color="#2AA1B3">**</font><font color="#A2734C">   Fw Ver: </font>MrChromebox-4.19.1 (01/29/2023)
<font color="#2AA1B3">**</font><font color="#A2734C">    Fw WP: </font>Disabled
<font color="#2AA1B3">*********************************************************</font>
<font color="#2AA1B3">**</font><font color="#26A269"><b> [WP]</b></font><font color="#A2734C"><b> 1)</b></font><font color="#2AA1B3"><b> Install/Update UEFI (Full ROM) Firmware </b></font>
<font color="#2AA1B3">**</font><font color="#26A269"><b> [WP]</b></font><font color="#A2734C"><b> 2)</b></font><font color="#2AA1B3"><b> Restore Stock Firmware </b></font>
<font color="#2AA1B3">**</font><font color="#26A269"><b>     </b></font><font color="#A2734C"><b> C)</b></font><font color="#2AA1B3"><b> Clear UEFI NVRAM </b></font>
<font color="#2AA1B3">*********************************************************</font>
<font color="#A2734C">Select a menu option or</font>
<font color="#C01C28">R</font> to reboot  <font color="#C01C28">P</font> to poweroff  <font color="#C01C28">Q</font> to quit 
2

<font color="#26A269">Restore Stock Firmware</font>

<font color="#A2734C"><b>Standard disclaimer: flashing the firmware has the potential to</b></font>
<font color="#A2734C"><b>brick your device, requiring relatively inexpensive hardware and some</b></font>
<font color="#A2734C"><b>technical knowledge to recover.  You have been warned.</b></font>

Do you wish to continue? [y/N] y

Do you have a firmware backup file on USB? [y/N] 

<font color="#C01C28">Unfortunately I don&apos;t have a stock firmware available to download for &apos;BOBBA&apos; at this time.</font>

<font color="#A2734C"><b>Would you like to use one from a ChromeOS recovery image?</b></font>

<font color="#A2734C"><b>This will be a 2GB+ download and take a bit of time depending on your connection</b></font>

Download and extract firmware from a recovery image? [y/N] y
<font color="#A2734C"><b>Sit tight, this will take some time as recovery images are 2GB+</b></font>

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  4074  100  4074    0     0  19766      0 --:--:-- --:--:-- --:--:-- 19873
Downloading recovery image inventory...
Processing board bobba
Downloading recovery image
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1727M  100 1727M    0     0  67.6M      0  0:00:25  0:00:25 --:--:-- 57.6M
Decompressing recovery image
Extracting ROOT-A partition
Extracting chromeos-firmwareupdate
Extracting coreboot image
Extracted coreboot-Google_Bobba.11297.357.0.bin
<font color="#A2734C"><b>Stock firmware successfully extracted from ChromeOS recovery image</b></font>

<font color="#A2734C"><b>VPD extracted from current firmware</b></font>

<font color="#A2734C"><b>Merging VPD into recovery image firmware</b></font>

<font color="#C01C28">Error clearing software write-protect range; unable to restore stock firmware.</font>

Press [Enter] to return to the main menu.
</pre>

MrChromebox, 2023-02-03:
I've removed that part from restoring the stock firmware as it's not needed, so try again

thonkinator, 2023-02-03:
Thank you, that worked perfectly. I noticed that after rebooting, the HWID is now "BOBBA TEST 4516"? Is there a way to reset this?

MrChromebox, 2023-02-03:
sure, use the script option to set the HWID. But you need to know the original HWID, or at the least a valid one for BOBBA. This is why you're supposed to back up your stock firmware.

thonkinator, 2023-02-03:
alright, i'll ask around. thank you!

closed 2023-02-03 by thonkinator