MrChromebox/firmware #430: Further Regressed Audio in Lillipup / TGL - kernel panic

marcosilva85, 2023-02-11:
Follow-up of issue #397  (https://github.com/MrChromebox/firmware/issues/397#issuecomment-1334644830)

With update to 4.19, which stated that  audio on Tigerlake is fixed, I can now report that unfortunately is now far more problematic than 4.18 :/

The 4.19 ROM now seems to make visible the audio to kernel so the snd-sof-pci-intel-tgl driver module loads. However:
1. Module cannot make the audio chip load properly, therefore there isn't any sound, but this broken state of audio brings random freezes here and there on user-space (Fedora 37).
2. At every shutdown, the kernel goes into panic and does not fully-power-off / stays on until 5sec button hard-shutdown. 
3. _Not a regression_ - type-c still does not work (it has never yet worked anyhow even on LEGACY).

Tested in Fedora37 and kernel 6.1.8 and 6.1.9.
For those who are curious I can tell that LEGACY_ROM at least had audio working (if you can endure on every boot having to use the keyboard for manually selecting down the "tianocore" option).

Fyi @MrChromebox. Hoping that you could have an idea if this could be fixed so that Lillipup could go out of this "alpha-experience" to a more stable platform, as its a nice price/chassis and components combination.

Fisiu, 2023-02-11:
This issue was already solved upstream, see January 2023 commits: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/sound/soc/sof/sof-audio.c?h=v6.2-rc7
Kernel 6.2 will have that fixed.

MrChromebox, 2023-02-12:
sigh, really annoying when people attribute kernel issues to firmware

marcosilva85, 2023-02-12:
Sadly the other way around also happens :-(
Ok so have to wait for the 6.2 release. @Fisiu do you know which kernel was the last one to have proper audio state?

Regarding type-c, it still isn't working , is this also a kernel issue @MrChromebox ?

Fisiu, 2023-02-12:
@marcosilva85, I'm not sure, I use patched 6.1.10 atm. I also did some quick tests with 5.4.xx, and sound was working too. It broke with some 6.0.x. However 6.2 is around the corner ;-)

ignapk, 2023-02-12:
Now the fix has been backported to 6.1.11 https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=f6c0ebc10c0b988dcd175789b478edf81446f2b7

marcosilva85, 2023-02-16:
I can report that is possible to have working audio again with kernel 6.1.11, though the driver module at boot is not properly initiated and sound does not work until the module is manually restarted at least once.

Regarding type-c, it still isn't working , is this also a kernel issue @MrChromebox ?

ellyq, 2023-02-24:
USB-C issues on TGL/ADL are linked to EC (Embedded Controller) and power sequencing.

I'm looking into that, but I broke my SuzyQ and it will take a few weeks for new board to arrive from China.
With coreboot built from 'next' branch, USB-C is working if you boot the machine with USB-C device connected (can be hotplugged in OS).

Not AltDP mode yet though, it will require us to make an EC call from userspace to let USB muxer know to enable this functionality.
[image]

coolstar, 2023-04-06:
not a firmware issue. should be closed. /cc @MrChromebox

ellyq, 2023-04-06:
Agreed, power sequencing has been fixed in 4.19.1. Now we just need a way to inform EC about the type of USB functionality it should enable from the userspace.

It can be done using ectool for instance, but it's not pretty. Either way, it's not a problem within coreboot.

closed 2023-04-06 by MrChromebox

marcosilva85, 2023-05-21:
With the latest firmware 4.20 I can report that I can see devices connecting via the USB-C port. Sometimes it does not trigger though.... I believe its working properly when I boot the laptop with the USB-C device already inserted.

> Agreed, power sequencing has been fixed in 4.19.1. Now we just need a way to inform EC about the type of USB functionality it should enable from the userspace. It can be done using ectool for [...]

Regarding the display-out / USB-C Alt display mode, did anybody get it working? It does not work for me.
@ellyq  would your comment be related to this?

ellyq, 2023-05-21:
I did, currently testing a patch before sending it to upstream.

As a temporary workaround, change module loading order or execute the following on fresh boot (as root):
`rmmod cros-ec-typec;rmmod intel-pmc-mux;modprobe intel-pmc-mux;modprobe cros-ec-typec`

Not sure if necessary, but without it I observed notorious system hangs, so: `modprobe cros-ec-debugfs`

This workaround fixes USB4, random freezes and sleep. With modules loaded in this order EC doesn't panic anymore, it's actually more stable than windows.