MrChromebox/firmware #431: Caroline Samsung Chromebook Pro - No Audio In Linux

cwayne-777, 2023-02-19:
Installed Linux Cinnamon 21.1
Pulseaudio Volume Control indicates "Dummy Output"
I ran the following command.......

~$ inxi -Axxz
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio driver: snd_soc_skl v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:9d70
  Sound Server-1: ALSA v: k5.15.0-60-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes

It appears that the card is detected but Linux doesn't see it. I also ran Linux Cinnamon from a USB stick.  Same problem.  Any suggestions are appreciated. Thanks in advance!_

MrChromebox, 2023-02-19:
this isn't a firmware issue, it's an "Intel has done a poor job of making Skylake/Kabylake Chromebooks usable under mainline Linux" issue. There's some discussions on the chrultrabook Discord (Linux channel) on how to get things working but it's not simple/straightforward yet which is why there's no how-to

closed 2023-02-19 by MrChromebox

MallocArray, 2024-05-27:
Shoot, I came here for the same reason, but looks like the Discord is no more and the current forum doesn't have any steps for resolving Audio issues with Caroline

MrChromebox, 2024-05-27:
there are no specific steps for any boards, there is a single script (maintained by the chrultrabook folks) which sets up audio under Linux for all Chromebooks as needed. Skylake boards like Caroline are supported. 

See https://docs.chrultrabook.com/docs/installing/post-install.html#linux

MallocArray, 2024-05-27:
Thank you so much @MrChromebox. You just saved my Samsung Pro from going back to stock firmware and running EOL ChromeOS
All I really wanted was to use ChromeOS Flex but with no audio, I tried Ubuntu and Pop_OS all without audio. 
This script got audio in Pop_OS working just fine, so I'm good to.  Not sure if I'll try it in ChromeOS Flex, it looks like it will not add audio there.

MrChromebox, 2024-05-27:
@MallocArray Flex like regular ChromeOS uses an immutable filesystem, so you'd need to enable making changes (ie, disable rootfs verification), but even then the audio script is set up for a standard linux audio subsystem, not ChromeOS' (cras)