MrChromebox/firmware #432: Usb won't mount in boot menu & EFI shell show blank

Achrafdif, 2023-02-20:
I mounted a usb flash drive with windows 10 burnt using rufus on it, then i restarted my chromebook, when i open boot menu, i don't see flash drive in boot menu nor in device manager, but i see it in boot maintenance manager, and when i try to boot from file doesn't do anything.
& when i tried opening EFI Internal shell it just shows a black screen backled.

MrChromebox, 2023-02-20:
* boot to UEFI main menu
* remove and reinsert USB
* go to boot menu
* select USB to boot

if that doesn't work, it's simply an issue with the USB stick itself not being compatible with the firmware.

When writing the ISO to USB with Rufus, be sure to use the defaults for GPT for UEFI. Change nothing else.

Achrafdif, 2023-02-20:
I've found someone with the same chromebook and the same issue & he solved it by updating from 4.19 to 4.12 but i don't know how to update plz how can i do that?!

MrChromebox, 2023-02-20:
you haven't even stated what Chromebook or firmware version you are using.

Achrafdif, 2023-02-20:
Acer gnawty cb3 111, firmware 4.19

MrChromebox, 2023-02-20:
did you try the instructions I posted above? There's no issue on the Baytrail devices I've tested here. Issues with USB sticks are not uncommon, sometimes you just need to use a different one

closed 2023-02-21 by MrChromebox

MrQOU, 2023-04-22:
Thanks for testing.
I can boot from the M2SSD, so it's not a big issue.
Thanks to the great firmware developers!

Wreckz91, 2023-06-16:
So I’ve tried multiple usbs I even have Ubuntu booted on live drive but can’t get it to boot using a brasswell its and Acer r11 cb5-132t

MrChromebox, 2023-06-16:
@Wreckz91 I'll take a closer look, but no issues on the left port here using the process outlined in my initial response above

Wreckz91, 2023-06-16:
[image]
That’s what I get when I boot and when I go into menu usb is not a boot option maybe I messed something up upon initial flash

MrChromebox, 2023-06-16:
type exit and press enter, like it says. then follow the steps I listed at the top above

Wreckz91, 2023-06-16:
> * boot to UEFI main menu
> * remove and reinsert USB
> * go to boot menu
> * select USB to boot
>
> if that doesn't work, it's simply an issue with the USB stick itself not being compatible with the [...]

Used Rufus and everything still nothing I’m going to try running as dd

joshwahls, 2023-06-20:
I have the exact same issue on the exact same model of Chromebook.

mnjenda, 2023-06-20:
This seems to be quite an issue with Acer chromebooks, I bricked two of my Acer C738t Chromebooks yesterday, to tied everything from Mr Chromebox's previous advices and still no show from the USB. And no way to go backwards it forwards regards installing windows or reverting back to Linux .
At first I thought it was the ventoy I was using on the USB, I tried Rufus and still no USB I'm dead

joshwahls, 2023-06-20:
> This seems to be quite an issue with Acer chromebooks, I bricked two of my Acer C738t Chromebooks yesterday, to tied everything from Mr Chromebox's previous advices and still no show from the USB. [...]

My sd card reader is showing in the boot list and appears like it may be working, I'm going to give that a try in a couple days.

mnjenda, 2023-06-20:
> I have the exact same issue on the exact same model of Chromebook. Acer r11 cb 738t
> 
> Two separate USB's from different companies, both ports on the chromebook, followed all of the instructions it never picks up the drive.
> 
> @MrChromebox please do attend

mnjenda, 2023-06-20:
> > This seems to be quite an issue with Acer chromebooks, I bricked two of my Acer C738t Chromebooks yesterday, to tied everything from Mr Chromebox's previous advices and still no show from the USB. And no way to go backwards it forwards regards installing windows or reverting back to Linux . At first I thought it was the ventoy I was using on the USB, I tried Rufus and still no USB I'm dead
> 
> My onboard sd card reader is showing in the boot list and appears like it may be working, I'm going to give that a try in a couple days.

mnjenda, 2023-06-20:
@joshwahls thanks a lot, hope it helps me and others too

MrChromebox, 2023-06-20:
> > I have the exact same issue on the exact same model of Chromebook. Acer r11 cb 738t
> > Two separate USB's from different companies, both ports on the chromebook, followed all of the instructions it never picks up the drive.
> > @MrChromebox please do attend

not sure what else I can do, I can't reproduce the issue here

MrChromebox, 2023-06-22:
any CYAN users who want to try a test build which may resolve the issue, please email me or msg me on Discord

tpowa, 2023-07-18:
Wrote you an email. Looking forward to test your fix.

MrChromebox, 2023-07-18:
@tpowa the current release is all I have available, there is no test build currently

MrChromebox, 2023-07-21:
USB detection should be improved in the just-dropped MrChromebox-4.20.1 release

haseebspaal, 2023-07-22:
> I have the exact same issue on the exact same model of Chromebook. Acer r11 cb5-132t
> 
> Two separate USB's from different companies, both ports on the chromebook, followed all of the instructions it never picks up the drive.
> 
> @Wreckz91 did you figure anything out?

i have a solution for this issue

tpowa, 2023-07-22:
> USB detection should be improved in the just-dropped MrChromebox-4.20.1 release
Fixed the issue for me. Thank you very much.

UNTUNGKSR, 2024-01-22:
Hello, I have the same issue here. Lenovo Chromebook C340-15 360, i3-8130 w /126eMMC. Detect as Yoga C630 Pantheon device in mrchromebox rec.
After following all the tutorials from mrchromebox w newest firmware 4.22.1, on the 1st try, I succeeded in booting the Windows Inst method, but I couldn't install it because there was not enough space. And then I cleaned up all the Chromebook partitions w/out a backup of it partition or the recovery. Then I rebooted my Chromebook, and this issue appeared, it can't recognize 64bit UEFI OS. Tried with Rufus and Ventoy but still, it doesn't work. Restart again to boot from the USB drive, from the boot file, nothing. But if I tried that USB on a Windows Machine, it worked normally both of them successfully booted without any problems.
So, please if any solution for this issue. Thx a lot mate.