MrChromebox/firmware #434: HP Chromebook 14 SNAPPY ChromeOS

Alek9nt, 2023-02-25:
Good day.

I just need help with this HP Chromebook 14 G5
It cant pass from recovery mode, reason no Bootable kernel on disk
it was turnt to UEFI firmware and then back to Stock
but it cant install the Chrome Os in recovery 
it always stuck: an unexpected error has occurred
but i'm not sure which model of chrome is, cause in recovery it say

Model: SNAPPYTEST 1088

I have try with some snappy models but no good recovery always fails.
the firmware Id is: Google_snappy . 9042 . 253 . 0
Chromebook model HP 14-ca061dx

any help with this, ill appreciate.

MrChromebox, 2023-02-25:
you need to perform a successful recovery for ChromeOS to be reloaded and bootable. The recovery media will have a logs telling you why it failed.

Alek9nt, 2023-02-27:
> 

Where i can find the Logs of the error.
The Chromebook has no system, and every time it fails stays in 
an unexpected error has occurred. please 
refer to this URL for troubleshooting tips.

i have left it for hours and never pass from there.

MrChromebox, 2023-02-27:
> Where i can find the Logs of the error.

on the recovery media (USB), as I already stated.

> i have left it for hours and never pass from there.

nor will it. You need to use the log to determine why recovery is failing

Alek9nt, 2023-03-21:
Good day

i want to know if there is a way to flash this Chromebook HP 14 G5 by it's Tipe-C usb via Fastboot

I can't find a solution to this. cant install the ChromeOS because the model is wrong

" Model: SNAPPYTEST 1088 " does not exist.

MrChromebox, 2023-03-21:
> i want to know if there is a way to flash this Chromebook HP 14 G5 by it's Tipe-C usb via Fastboot

I provide info on how to do this in my unbricking guide - https://wiki.mrchromebox.tech/Unbricking, but I'm not sure that's the issue here?

> " Model: SNAPPYTEST 1088 " does not exist.

what is reporting that error? There should be a space between 'SNAPPY' and 'TEST' - is there?

MrChromebox, 2023-03-22:
you just need the SNAPPY image, what comes after shouldn't matter. Or just select one of the devices listed as SNAPPY on my site from the 'select a model from the list' function

Alek9nt, 2023-03-22:
I already tried, with different snappy models, many times
Tried with USB, SD different memories just to make sure but nothing
Can't make it work

I thought it has to be the firmware installed, cause i never have this much trouble before, besides this one is.... kind of slow while loading the recovery process

MrChromebox, 2023-03-22:
and I can't help you with a failed recovery if you can't produce the recovery log. Trying another image isn't going to fix the problem

Alek9nt, 2023-03-22:
It does make the log files, but i don't really understand the error

[recovery.log](https://github.com/MrChromebox/firmware/files/11043002/recovery.log)

MrChromebox, 2023-03-22:
> 2020-07-25T06:15:57.448440Z INFO cros_installer: [chromeos_setimage.cc(109)] Setting up verity.
> 2020-07-25T06:16:09.403997Z ERROR cros_installer: [chromeos_verity.cc(269)] Filesystem hash verification failed
> 2020-07-25T06:16:09.404078Z ERROR cros_installer: [chromeos_verity.cc(270)] Expected c119a37e0dadb6bf7ccd5fb9a1a76c045222297240f336ce22bd0f3982958159 != actual ca19c18e0b21049ae8de7720dba4bf9a946d5fd00c4647c954b4c2538ab32b4f

so, it's copied the ChromeOS image to the eMMC, but (part of) it fails verification. So either an issue with the USB stick or your eMMC

MrChromebox, 2023-03-23:
this isn't something I can help with, it's not a firmware issue. If the Chromebook is new, make use of the warranty on it

closed 2023-03-23 by MrChromebox