MrChromebox/firmware #435: HP Chromebox TPN-Q150 (Zako model) developer mode disabled

JammerJMC, 2023-02-27:
I recently purchased several, used HP Chromebox TPN-Q150's on Ebay. My plan is to use MrChromebox's firmware utility tool to install linux onto the hard drives. Unfortunately, a couple of them have developer mode disabled, which has me dumbfounded seeing as I was under the assumption that all one needs to do to access developer mode on Zako Model chromeboxes are to remove the inner screw, which I have done. I have successfully modified several of this model in the past, which is why I'm a bit confused as to why they wont work. Any input would be greatly appreciated.

MrChromebox, 2023-02-27:
the firmware write protect screw has nothing to do with Developer Mode.

If the device is managed/enrolled, there's no way to bypass to enable developer mode if locked out. Best option is to flash firmware chip directly (treating the device as bricked, essentially).

If the device was previously managed/enrolled but is not anymore, developer mode may remain locked out until one logs in with a non-guest account to establish a new device owner.

JammerJMC, 2023-02-28:
> the firmware write protect screw has nothing to do with Developer Mode.
>
> If the device is managed/enrolled, there's no way to bypass to enable developer mode if locked out. Best option is to flash [...]

Thank you so much for your reply. how would one go about "flashing the firmware chip directly"? Could one do it them self or is there  special tool that is needed? I don't mind diving into it so long as I have some guidelines on how to so, as well? Is there an online walkthrough of the process?

JammerJMC, 2023-02-28:
> the firmware write protect screw has nothing to do with Developer Mode.
>
> If the device is managed/enrolled, there's no way to bypass to enable developer mode if locked out. Best option is to flash [...]

Thanks again for your input. I was able to Google how to "change ownership" of Chromebox and found a tutorial on how to do so as well as reset the device to factory defaults. Once, I did that, I was able to enter into developer mode & use the firmware utility tool.

closed 2023-02-28 by MrChromebox

JammerJMC, 2023-03-01:
I do have another question. I was wondering what is the best firmware script to use while in Developer mode in order to access the firmware? I have carefully entered both firmware utility scripts, making sure that they are entered exactly as they appear on the mrchromebox.tech website and neither seem to be working. any input would be greatly appreciated.

MrChromebox, 2023-03-01:
there's only one script, so not sure what you're asking. I can't help with "don't seem to be working" - please post the exact error, or better yet a pic

JammerJMC, 2023-03-01:
[image]