MrChromebox/firmware #436: Firmware utility script down or not working?

JammerJMC, 2023-03-01:
I was wondering what is the best firmware script to use while in Developer mode in order to access the firmware? I have carefully entered both firmware utility scripts, making sure that they are entered exactly as they appear on the mrchromebox.tech website and neither seem to be working. I did a little research on Youtube and have stumbled across a method to use the kodi version. However, neither of them appear to be working either.  Any input would be greatly appreciated.

MrChromebox, 2023-03-01:
> seem to be working

this is not diagnostically useful. post a pic so I can see exactly what you are entering (and where) and what the output/error is

JammerJMC, 2023-03-01:
[image]
 Here's a pic. I apologize for the glare. However, the text is visable. Am I somehow in a wrong directory?

JammerJMC, 2023-03-01:
> you're...not connected to the internet.
> 
> the only command you should be using is the firmware-util.sh one. I need to deprecate the Kodi one as I've not touched it in years

Thanks so much for your input. I do have an ethernet cable plugged in. However, after reading your post, I checked it and it was a little loose, which was not apparent by simply looking at it.

MrChromebox, 2023-03-01:
"Cannot resolve host" is a dead giveaway for DNS failure, which usually means you're not connected to the internet :)

closed 2023-03-02 by MrChromebox