MrChromebox/firmware #437: Old HP Chromebook 14 won't boot or take recovery media

artisancreek, 2023-03-02:
I have an HP Chromebook 14-ca043cl and it won't boot/turn on. So I pressed the Esc+Recovery+Power key to get the recovery screen. The screen did come up and provided "SNAPPY E25-C3C-32Y-A7Z-U3Q-A3D". In trying to use the Chrome extention to create the recovery media, my model isn't quite there. I tried a few HP Chromebook 14 (G3, G7, db0000-db0999/14A G5, q000-q099 WP2 / HP Chromebook 14-SMB Intel Corp) and when I put the media into the Chromebook, I get "The device you inserted does not contain Chrome OS"

1. Which model should I be choosing?
2. Why isn't it recognizing the media (brand new, bought just for this)
3. Suggestions?

Thanks

MrChromebox, 2023-03-02:
this has nothing to do with ChromeOS firmware...

> In trying to use the Chrome extention to create the recovery media, my model isn't quite there

just put in SNAPPY and it will select the correct image

artisancreek, 2023-03-02:
Hmmm...  I put in SNAPPY, and it responds "No model found yet. Keep typing" And the "Continue" button is not enabled. 
I did try and type more from the screen, but it stopped.

MrChromebox, 2023-03-02:
> I did try and type more from the screen, but it stopped.

what does that mean? what did you type? the rest of the HWID? what stopped?

artisancreek, 2023-03-02:
Basically said too many characters - did it about 3-4 times.  BUT, because you seemed surprised, I tried it once more, and it found a match - go figure.
It is starting recovery.  Thank you so much for your help.

closed 2023-03-02 by MrChromebox