MrChromebox/firmware #438: Wacom emr stylus not working (Akali 360)

littlepluma, 2023-03-06:
Device: Acer Chromebook Spin 13 (Akali 360) 
Firmware Version: 4.19 (UEFI full rom)
OS:  Ubuntu 20.04/22.04(xorg)

I have read about similar issues which are closed by now and learnt that the stylus problem should be fixed in 4.19 firmware. However it still seems to be existing in my case. 
By the way, is there any approach to fix the sound problem?  It is nothing major to me, just asking. 

`yizhitai@yizhitai-Akali-360:~$ lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 08)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.1 USB controller: Intel Corporation Device 9d30 (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:15.2 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #2 (rev 21)
00:15.3 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #3 (rev 21)
00:19.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #2 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #4 (rev f1)
00:1e.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #0 (rev 21)
00:1e.2 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO SPI Controller #0 (rev 21)
00:1e.3 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO SPI Controller #1 (rev 21)
00:1e.4 SD Host controller: Intel Corporation Device 9d2b (rev 21)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC Controller/eSPI Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Multimedia audio controller: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.5 Non-VGA unclassified device: Intel Corporation Device 9d24 (rev 21)
01:00.0 Network controller: Intel Corporation Wireless 7265 (rev 59)

yizhitai@yizhitai-Akali-360:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b667 Chicony Electronics Co., Ltd HD User Facing
Bus 001 Device 002: ID 8087:0a2a Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

yizhitai@yizhitai-Akali-360:~$ xinput 
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Elan Touchscreen                        	id=9	[slave  pointer  (2)]
⎜   ↳ Elan Touchpad                           	id=10	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ HD User Facing: HD User Facing          	id=8	[slave  keyboard (3)]
    ↳ Intel Virtual Buttons                   	id=11	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=12	[slave  keyboard (3)]
`

littlepluma, 2023-03-06:
and the dmidecode result
`# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.
Table at 0x7AA2D000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: coreboot
	Version: MrChromebox-4.19.1
	Release Date: 01/29/2023
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 4.19
	Firmware Revision: 0.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Google
	Product Name: Akali 360
	Version: 1.0
	Serial Number: N9EFJWW004828146C27600
	UUID: Not Settable
	Wake-up Type: Reserved
	SKU Number: sku11239
	Family: Google_Nami

Handle 0x0002, DMI type 2, 14 bytes
Base Board Information
	Manufacturer: Google
	Product Name: Akali 360
	Version: 1.0
	Serial Number: 827000WHMBQC00005
	Asset Tag: Not Specified
	Features: None
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Google
	Type: Laptop
	Lock: Not Present
	Version: Not Specified
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Not Specified

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
	Socket Designation: CPU0
	Type: Central Processor
	Family: Pentium Pro
	Manufacturer: GenuineIntel
	ID: EA 06 08 00 FF FB EB BF
[... 207 lines cut ...]
Onboard Device
	Reference Designation: VGA compatible controller
	Type: Video
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:02.0

Handle 0x0011, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Multimedia audio controller
	Type: Sound
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:1f.3

Handle 0x0012, DMI type 133, 5 bytes
OEM-specific Type
	Header and Data:
		85 05 12 00 01
	Strings:
		KHOIHGIUCCHHII

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table
`

MrChromebox, 2023-03-06:
please provide a cbmem console log, a DMI dump tells me nothing useful

littlepluma, 2023-03-06:
[cbmem.log](https://github.com/MrChromebox/firmware/files/10903726/cbmem.log)
:-)

MrChromebox, 2023-03-07:
looks like the reset GPIO for the stylus is getting held low, so not being detected. drop me an email and I'll provide a test build which should have the issue fixed

littlepluma, 2023-03-07:
yizhitai@outlook.com
Thanks a lot

littlepluma, 2023-03-07:
After installing the test build the stylus works perfect.
Thanks again :-)

closed 2023-03-07 by littlepluma

krrrak, 2023-05-03:
same problem.may I get the test build?

MrChromebox, 2023-05-03:
@krrrak the coreboot 4.20-based release should be happening fairly soon, so just be on the lookout for that