MrChromebox/firmware #440: [SUCCESS] I have just unbroken AltFw on Alder Lake

BlinD-HuNTeR, 2023-03-08:
It has been known for some time that AltFw support in Depthcharge seems to be broken on some newer Tiger Lake and Alder Lake boards, as noted in issue #122. Apparently on those platforms, trying to flash any payload in RW_LEGACY just gives an error message when trying to boot it, pointing to the firmware log for more info.

I tried this on my machine, an Asus Chromebook Flip CX5601 (Alder Lake, board is Felwinter). I checked the firmware log, and the error was `segment type 0x45444F43 not implemented`. This is an error message from file [payload.c](https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/platform/depthcharge/src/boot/payload.c;l=135) from Depthcharge source code. But it makes no sense, because `0x45444F43` is `PAYLOAD_SEGMENT_CODE`, which is clearly covered in the switch case.

After more investigation I just discovered that this code is being compiled with wrong endianness. In coreboot's source tree there is a `cbfs.h` file and a `cbfs_serialized.h` file. They both define the same constants, but with reversed byte order, **and Depthcharge is using the wrong one**.

This means that the current Depthcharge version, which is built into many chromebooks, **is expecting payloads with wrong segment types**. To work around that, I just needed to compile cbfstool with these modified constants. Cbfstool uses `cbfs_serialized.h`, so I reversed the byte order on all 5 `PAYLOAD_SEGMENT` constants in this file and compiled it.

After doing that, I went on with trying some payloads. I tested first with GRUB compiled for coreboot, and it loaded just fine from the AltFw menu. Then I compiled your fork of Tianocore, branch upp_202301, and tried to boot it, but the screen just froze. I'm guessing it might have something to do with graphics, cause I pressed Ctrl-Alt-Del on an external keyboard and it just rebooted, which means that the machine was not hung.

But finally, instead of compiling Tianocore myself, I decided to try a prebuilt one. I checked your `sources.sh` file, and found that the most recent RWL rom was `rwl_altfw_glk-mrchromebox_20230110.bin`. Despite the name clearly saying "Gemini Lake", I understand that Tianocore is more or less platform agnostic, so I extracted the payload from it, flashed it to my RW_LEGACY, and it booted!

As the last step, I tried to boot Windows from Tianocore, which is my main goal after all. I have read a lot and I know that Windows may have lots of ACPI problems if you don't use the full firmware, but since it's not available for Felwinter board yet, I guess I have no choice for now.

So I booted, and I was expecting to get some ACPI error, but instead got a BSOD with code `MULTIPROCESSOR_CONFIGURATION_NOT_SUPPORTED`. Not sure what this means, but it might be because I tried to boot an existing Windows install, that was installed on QEMU. I still didn't try to boot the Windows installer and make a fresh install.

Anyway, if you use this custom build of cbfstool, you will be able to support a lot more platforms with the RW_LEGACY firmware. Even if this may not be your main goal, it's still quite useful for lots of people, as it has 0 bricking risk and doesn't require disassembly or debug cable.

Thank you so much for your work on this project! Also if you can point me on the direction on how to get Windows to boot in my scenario (or perhaps even help me with the full firmware on my board), I'd really appreciate that!

coolstar, 2023-03-09:
Hey there! If you'd be happy to hop into the chrultrabook discord, I'd be happy to build you a felwinter full ROM image to test Windows (banshee and osiris have full driver support, and the same drivers should work on felwinter too).

Since you did manage to fix RW_LEGACY, that is a potential option too; though keep in mind that you'll need to chainload OpenCore on top of tianocore to fix up ACPI (like how Ryzen chromebooks boot Windows via RWL) which is a bit clunky and cumbersome. This is do-able, but since you're the first, the patches will need to be generated. See https://coolstar.org/chromebook/windows-rwl/install-rwl.sh for how the patches on Ryzen are done (though obviously different patches needed for Brya).

That being said, you may want to disable write protect anyways to set GBB flags and update EC firmware because the stock one has a bug that makes Windows take 10 seconds longer to boot w/ the built in keyboard driver.

happybrahma, 2023-05-09:
I have an alder lake OSIRIS board chromebook.  Acer 516 GE chromebook.

I'd be happy to test out an alt firmware binary if anybody has built one..  I'm sure I can figure out how to flash it.  I'd even be willing to build it myself with some noob-friendly instructions.  I have a 20 year development background, but I have  never really messed around much with this platform, so I've got a little learning curve to get through.

happybrahma, 2023-05-09:
Probably worth mentioning that I'm just trying to boot grub/ubuntu LTS 22.04.2

closed 2023-05-09 by MrChromebox