MrChromebox/firmware #444: Boot freezes on splashscreen if specific harddrive attached

hydriniumh2, 2023-03-19:
Firmware: 4.19.1, full ROM
OS: OpenMediaVault 6.3.4-1/Debian 11.6
Hardware: Chromebox 3 CN65, i7-8550U 

Hi, 
First, thanks for maintaining and writing this amazing resource, it's really nice being able to use the OS of my choice on a Chromebox!
That being said I've been having an odd issue where if I plug in a specific harddrive (Touro Mobile 2 TB) the chromebox won't boot. I was hoping to use the chromebox as a NAS so I have several USB harddrives attached to it, unfortunately it seems like something about the Touro is messing with the UEFI bios startup. I have 4 other harddrives attached as well but they all work fine so it seems to be specific to this one harddrive. 

Behavior:
- Plug in Touro Harddrive
- Start Chromebox
- Boot will hang indefinitely on Rabbit splashcreen with no "Press ESC to enter bios options" text appearing.

Some troubleshooting steps I've already done:

- If I disconnect the harddrive and then re-attach it after startup the Touro 2TB works fine. 
- If I restart with the harddrive attached the bios will hang at the rabbit splashscreen
- If I start with the harddrive disconnected, then go into the UEFI options and reconnect the harddrive the menu will sometimes freeze
- The harddrive itself doesn't appear to be dying or anything, Crystaldiskmark comes back clean.

Also, I don't know if this is relevant but the bios doesn't have the "Boot Maintenance Manager" option, and also doesn't save options between reboots. Although there aren't really any options to speak of, just changing boot order.

Any insight/help would be greatly appreciated!

Thanks

MrChromebox, 2023-03-19:
have you tested with a 95W barrel-jack PSU instead of USB-C?

If no difference there, it's simply a bug in the USB init of the UEFI firmware, which would require a Suzy-Q cable and debug build to try and resolve

> Also, I don't know if this is relevant but the bios doesn't have the "Boot Maintenance Manager" option

it was renamed to Boot Manager a while back

hydriniumh2, 2023-03-19:
Technically no, but I'm using a [barrel plug USB-C pd adapter](https://www.adafruit.com/product/5452) because I wanted to keep the USB-C port open to use USB-C DP output. 
For the debugging I'm seeing that the only option for the Suzy-Q cable is DIY due to supply chain issues, I can order the breakout and make the cable myself but I'm not finding much in terms of instructions on how to use the cable/ what steps to take to debug the UEFI.
Before I attempt to try and fix the issue could you give a broad estimate on how involved the process would be? If fixing the issue has a low chance of success I'd probably just KISS and just not use the harddrive.

Also, would you have any speculation on why just 1 brand/harddrive is causing this issue?
Thanks!

MrChromebox, 2023-03-19:
> Before I attempt to try and fix the issue could you give a broad estimate on how involved the process would be? If fixing the issue has a low chance of success I'd probably just KISS and just not use the harddrive.

build Suzy-Q cable, compile coreboot + Tianocore with serial output and debug build enabled, record/log a boot using the Suzy-Q, then parse the log to try and determine what's causing the issue. Narrow it down, add some more debug output, rinse and repeat.

KISS is going to be a lot easier. Maybe see if the drive controller itself has a firmware update.

> Also, would you have any speculation on why just 1 brand/harddrive is causing this issue? 

it could have buggy firmware, or the Chromebox could have buggy firmware

hydriniumh2, 2023-03-20:
Just wanted to give an update, I figured out that the issue was more related to my OpenMediaVault installation + harddrive + powered USB hub interacting than the firmware. 

I'm still not sure what's causing the problem, but part of the issue was that the drives are on a powered USB hub so they don't fully shut off even after the machine powers down. The other 4 drives are fine with this but specifically the weird harddrive if you don't fully power cycle it between reboot it will hang the OS. 

Very strange behavior, couldn't really think of any reason why this would be because I've never seen the drive do this on any other computer, even another OMV installation on a different machine.
Anyways, the solution was to simply plug that one harddrive directly into the USB ports of the Chromebox so that it fully shuts off between reboots.
Closing the issue as resolved and thank you for answering my question even though it turns out to have been mostly unrelated to the firmware.

closed 2023-03-20 by hydriniumh2