MrChromebox/firmware #446: Supported Linux Distro for running Firmware-util?

agnivo007, 2023-03-23:
I used  MX linux live iso for running firmware utility script, but the latest release seems to throw up error regards to missing glibc versions 2.33 and 2.34 ...

What is the best/supported Linux live distro to run current version firmware utility script without any tweaking?

MrChromebox, 2023-03-23:
probably a Ubuntu 22xx live ISO, I need to figure out a good way to support diff glibc versions

ninelore, 2023-03-24:
Ive made the experience that `picutils` > 3.8.0 has Issues with flashing, but that could be fixed by now.
Besides that you can go into edit mode of the live usb's boot loader and add the kernel parameter `iomem=relaxed` (had to do that on arch)

@MrChromebox about glibc: You should select the oldest distro that you want to target and build on that one. Debian bullseye  or Ubuntu 20.04 LTS (both are on 2.31) would be my suggestion.

ninelore, 2023-03-26:
@MrChromebox what do you think about shipping `pciutils <= 3.8.0` for now?

MrChromebox, 2023-03-26:
@ninelore I am - 3.7.0-6 is what I'm compiling flashrom against

SamT888, 2023-08-20:
> I used MX linux live iso for running firmware utility script, but the latest release seems to throw up error regards to missing glibc versions 2.33 and 2.34 ...
> 
> What is the best/supported Linux live distro to run current version firmware utility script without any tweaking?

FydeOS works as well

MrChromebox, 2023-10-12:
closing as this is a script issue not a firmware one

closed 2023-10-12 by MrChromebox