MrChromebox/firmware #447: Lenovo Flex 5i (LILLIPUP) Unable to boot from eMMC with UEFI 4.19

TheFeshy, 2023-03-28:
After following the instructions to flash the UEFI, and installing Debian 11.6, rebooting gives me the message:

`Booting from 'debian' failed; verify it contains a 64-bit UEFI OS.

Press any key to continue booting...`

This is identical to the problem discussed at the bottom of the closed issue #397, but since that issue provided the whole UEFI image, I thought it might be easier to track as a new issue for this smaller problem.

I can confirm that this device does not show up on the EFI shell's map.  The bootloader is looking for the path `HD(3,GPT,5D708219-005C-D745-99FA-AC3F7D6E5B69,0X800,0X80000)/EFI/Debian/shimx64.efi` but not only is this not found in the map, but nothing is.  If I leave my install USB thumb drive in, the shell will detect those and populate the map (fs0:, fs1:, etc.)  But there is no sign of the eMMC, and if I leave the USB stick unplugged, the map is completely empty.

My model of LILLIPUP has the 64GiB eMMC, and it appears to be a 2242 PCIe card, like an NVMe.  It's driven by a GL9763E chip, which seems to be a PCI to eMMC controller?  I can see that Coreboot should have support for this chip (there is code for it [in their repo](https://elixir.bootlin.com/coreboot/4.19/source/src/drivers/genesyslogic/gl9763e/gl9763e.c)), but the drive is not found at boot.

I should also say that linux kernels 5.18 and later _also_ fail to handle this drive properly - I get the message `PLL clock never stabilised.` followed by other error messages depending on linux version.  5.18 is the kernel version that added this message, with [this](https://lkml.org/lkml/2022/3/7/188) patch.

Issues with two different drivers (UEFI and linux) would normally make me think hardware failure - but the drive worked fine in chromeos, and it works fine in Debian (still on kernel 5.10) 

Long term, the answer for me is to replace it with an NVME (hopefully the slot is properly plumbed for that), but I also wanted to document it here in case anyone else either had the same problem or had found a solution.

TheFeshy, 2023-03-29:
Replaced the eMMC with a 2242 NVMe, and everything worked as expected.

osousa, 2024-01-06:
> Replaced the eMMC with a 2242 NVMe, and everything worked as expected.

Hi there!

I have the same model, with NVMe, does everything work properly on Lenovo Flex 5i (LILLIPUP) , using UEFI Full ROM? 

Thanks!

TheFeshy, 2024-01-08:
The only other issue I have had with it is sound not properly working after suspend.  This can usually, but not always, be fixed by unloading and reloading the sound mod from the kernel (snd_sof_pci_intel_tgl)

Otherwise I've been really happy with how things have "just worked" - even the third-party pen I bought for the tablet interface.

osousa, 2024-01-28:
> The only other issue I have had with it is sound not properly working after suspend. This can usually, but not always, be fixed by unloading and reloading the sound mod from the kernel [...]

Thank you very much for your reply, and i am sorry for the late response, only today i had the time to go through the process.

Everything went smoothly, booted up from USB, tried the distro and all ,everything works beautifully from sound to video, to touchscreen

But when i went to install it, i noticed it tried to install it on the USB stick from where its booting... it simply didn't detect the SSD (Union Memory M.2 NVMe , sounds like a OEM brand)

I tried running "map" command on UEFI shell, nothing shows up... i think you ran "map" as well, but our situation differs in the sense that the OS at least could "see" the disk and installed the OS on it , right? 

I tried everything while running the live linux to try and find the M.2 SSD, nothing worked.

With respect to the UEFI boot loader, i know too little to try and do something other than running basic commads like "map" :(

I took note on the disk you purchased, just wanted to get your 2cents before spending any more money on this endeavour of having linux on a cheaper laptop . Thank you very much.

TheFeshy, 2024-01-30:
I see your edit - I wonder if it has the same eMMC controller that caused me problems?  Although in my case, as you say, it was visible under linux - just not the coreboot UEFI.  I'm not sure what the general compatibility is for eMMC in linux; I avoid them since the price difference is usually so low that it isn't worth the performance hit.

The model of NVME I bought is Samsung MZALQ256B 256GB, if that helps.  It's form factor is 2242 M key.

osousa, 2024-02-09:
> The model of NVME I bought is Samsung MZALQ256B 256GB, if that helps. It's form factor is 2242 M key.

Ah! Too bad i decided going for the cheap route and acquiring a Kingspec for $25 ... i don't know if its related, but installed Ubuntu 23 and it started freezing a few minutes after using it.  Completely freezes and needs reboot.

Tried changing the distro to Debian 12, the issue seems to not be as recurrent but it freezes sometimes too :( 

Stress-tested the RAM , no issues, i recognise this won't be an easy issue to fix. Wondering if its worth it spending another $50 for a quality NVMe just to hit this wall yet again...

But at least i was able to install and run it. Thanks!

closed 2026-11-04 by MrChromebox