MrChromebox/firmware #448: Which coreboot to use for CTL CBX1-7 (wukong)?

jeremyhuytran, 2023-03-29:
I am seeing this model is supported on the table but while trying to flash the bios chip with coreboot, I could not find any wukong coreboot available. Is it missing or I can use other to substitute? 
This box has intel core i7 8550u

MrChromebox, 2023-03-29:
well, what are you trying to do with the box?

If you want to run ChromeOS, then you just need to edit the serial in the RO_VPD, clear the RW_VPD, and perform a recovery.

If you want to run Windows/Linux, then flash the UEFI firmware (fizz image) as per my Unbricking guide, taking care to preserve the RO_VPD so the LAN MAC doesn't get erased

jeremyhuytran, 2023-03-29:
Thanks for that info.  
1. If i still want to use the google chromeos, you said we need to read and change the RO_VPD, then  clear the RW_VPD. I believe we use the cbftool and flashrom to do the job. Do you have a list of command for this somewhere? 

2. for installing window/linux

I did follow your unbricking instructions for my other box as the box has no screen on boot.  For the step "Determine file to be flashed", i wget both coreboot.rom and shellball.bin for panther board. Then, for the later steps ( extract and inject vpd,  and final flash write ), I only used the coreboot.rom as filename input.
So, i am not sure if the shellball rom would play any role in my doing or it is just an alternative rom option type  for other specific purposes. Can you clarify that?

Sorry that i have 2 long paragraphs as questions. I have been buying many chromebox in the last year for really cheap ($20 or less each) and your project made me so excited

MrChromebox, 2023-03-29:
you don't need a shellball rom, period.

1) read flash 

2) extract RO_VPD using cbfstool

3) doesn't matter. "0000000000" "1234567890" "HI_MOM"

4) use google vpd tool to change so checksum is correct. https://mrchromebox.tech/files/util/vpd.tar.gz
`./vpd -f ro_vpd.bin -s "serial_number"="1234567890"`

5) `./vpd -f badflash.rom -O -i RW_VPD`
only applicable to stock firmware. use vpd to edit stock firmware directly if sticking with ChromeOS, no need to extract RO_VPD to file to edit

jeremyhuytran, 2023-03-29:
Thank you so much. I will try it tonight.

closed 2023-03-30 by jeremyhuytran