MrChromebox/firmware #450: thinkpad c13 yoga chromebook updated RW_legacy Bios,cant install Windows,got ACPI_BIOS_ERROR

AdorableGhost, 2023-03-31:
thinkpad c13 yoga chromebook updated RW_legacy Bios,cant install Windows,got ACPI_BIOS_ERROR.
the firmware is rwl_altfw_zen2-mrchromebox_20210623.bin ,tried many times that windows give me ACPI_BIOS_ERROR ,and ChromeOs version is 111.5563.100 .what's wrong with this.is the chromeOs version too high to install windows or I installed something wrong?

MrChromebox, 2023-03-31:
support for running windows is handled by coolstar on www.coolstar.org and the chrultrabook subreddit / discord server.

closed 2023-03-31 by MrChromebox

AdorableGhost, 2023-04-03:
but If followed the instructions from www.coolstar.org then I got the error ...and somebody else have been succeed .

AdorableGhost, 2023-04-03:
collstar said My chromeBook can install  Windows......

MrChromebox, 2023-04-03:
this isn't a problem I can fix, it's what coolstar's ACPI patches are for. 
I don't support running Windows on Chromebooks, coolstar does.