MrChromebox/firmware #452: Eldrid Full Rom Blinking Power LED

frolfer-in-a-tux, 2023-04-11:
I did a full rom flash on my device. It seemed fine until I left the device powered off overnight. When I open it up, the power led comes on and off twice then nothing. It stays powered off. When I connect the AC adapter, the power LED starts to flash continuously. I am able to recover it by disconnecting the battery and then connecting the AC adapter. Anyone experiencing this?

MrChromebox, 2023-04-12:
you shouldn't have to disconnect the battery, a hard reset (Refresh+Power) should do. You also haven't told us what firmware version, OS, etc you're using.

frolfer-in-a-tux, 2023-04-12:
Oh. I tried power plus refresh before but guess I didn't hold it long enough. Sweet. That saves me from scraping up the metal more when using a flathead to pry it open.

Now it's powered on. I am running firmware 4.19.1 and Fedora Silverblue 38 but it also did the same on 37.

MrChromebox, 2023-04-12:
IDK, I'd recommend asking on the chrultrabook Discord as there are others there with this device; I don't have one myself

closed 2023-04-19 by frolfer-in-a-tux