MrChromebox/firmware #457: Eldrid Full Rom USB Type C Support

frolfer-in-a-tux, 2023-04-19:
Full Rom 4.19.1
Fedora Silverblue 38

USB C ports, both sides, are not working. From bios I can only boot from usb drives using the type a port. From Fedora usb c devices are not recognized. dmesg shows no new messages after plugging in devices to the usb c ports. Same device using a type a to type c adapter are recognized on the type a port.

frolfer-in-a-tux, 2023-04-25:
To be more specific. Charging works on USB C ports. Data does not.

arek-neptuneweb, 2023-05-08:
I had the same problem with full rom 4.19.1 on debian testing. It seems to be fixed once I used the liquorix kernel.

frolfer-n-a-tux, 2023-05-22:
On 4.20 usb flash drives and external hard drives are now showing up on boot when connected to the usb c ports. The right port seems to support full booting. The left port will boot but then stop reading device.

For Fedora I am using a workaround that was shared in the chrultrabook discord. It involves unloading 2 kernel modules and then reloading them in a specific order.

rmmod cros_ec_typec
rmmod intel_pmc_mux
modprobe intel_pmc_mux
modprobe cros_ec_typec

MrChromebox, 2025-12-03:
this is an OS issue not a firmware one, should be resolved by now

closed 2025-12-03 by MrChromebox