MrChromebox/firmware #461: coreboot 4.19 cannot boot from USB flash drive.

MrQOU, 2023-04-22:
After updating chromebox cn62 to coreboot 4.19 I can't boot from USB flash drive (monitor is black screen).
Tested bootable USB flash drives are Win10/11, Ubuntu and LinuxMint.

However, the M.2 SSD can boot just fine.
Boot ubuntu with M.2 SSD and do NVRAM clear with firmware-util.sh, next time you can boot from USB flash drive only once.

Do you have any solution?

MrChromebox, 2023-04-22:
what board are you using?

MrQOU, 2023-04-22:
Core i7 5500U model. Is this what you want to hear?

thank you.

MrChromebox, 2023-04-22:
sorry I missed the device listed in the OP. I'm not able to reproduce here on my CN62, it boots from USB fine repeatedly

MrQOU, 2023-04-22:
Thanks for testing.
I can boot from the M2SSD, so it's not a big issue.
Thanks to the great firmware developers!

MrQOU, 2023-05-02:
> @MrQOU ブートメニューで、ファームウェアが追加されたドライブをブートシーケンスに保存しないことが原因です。USBから起動するたびに、`esc`起動メニューでドライブを押して選択する必要があります。

Thank you for your reply.
But even your procedure doesn't boot.
I verified it by the following steps.
(all Win10 installation USB)

[Boot Menu]->[USB Device]
don't boot

[Boot Manager]->[Boot From File]->[USBDevice]
bootmgr.efi
don't boot

[Boot Manager]->[Boot From File]->[USBDevice]
efi\boot\bootx64.efi
don't boot

[EFI Shell]
fs0:
efi\boot\bootx64.efi
don't boot

MrQOU, 2023-05-02:
> @MrQOU Do mean in the boot menu cause afaik the firmware doesn't save added drives to the boot sequence. Every time I want to boot from a usb i have to press `esc` and select the drive in boot menu.

Thank you for your reply.
But even your procedure doesn't boot.
I verified it by the following steps.
(all Win10 installation USB)

[Boot Menu]->[USB Device]
don't boot

[Boot Manager]->[Boot From File]->[USBDevice]
bootmgr.efi
don't boot

[Boot Manager]->[Boot From File]->[USBDevice]
efi\boot\bootx64.efi
don't boot

[EFI Shell]
fs0:
efi\boot\bootx64.efi
don't boot

MrChromebox, 2023-05-02:
what does "don't boot" mean? what's the exact error or system response?

MrQOU, 2023-05-02:
> what does "don't boot" mean? what's the exact error or system response?

i made a video Please watch.
https://photos.app.goo.gl/1M7o1Q44gFJRtWRu8

1.
Initialize NVRAM from ubuntu in M2SSD and shutdown.

2.
Boot into an EFI shell and boot the Win10 installation disk in your USB flash drive.

3.
I reboot and try the same. but it doesn't boot.
It finished loading and stopped responding.
The [Enter] key and [CTRL][ALT][DEL] do not respond.

MrChromebox, 2023-05-03:
I don't understand, why are you booting to the UEFI shell? You simply select the USB from the boot menu. period.

MrQOU, 2023-05-03:
I made a video to prove that the .efi file exists and is trying to boot.

As per your instructions, I chose the USB flash drive from the boot menu.
I share that video.

https://photos.app.goo.gl/zZqP8oQGGZdwBnq3A

I apologize for taking so much time.

MrChromebox, 2023-05-17:
please try with the just-release MrChromebox-4.20.0 firmware, it has an updated UEFI payload

MrQOU, 2023-05-17:
Thanks for letting me know about the new firmware.
i tried now.
Problem solved, great!

I will make a small donation. good luck to you!

MrChromebox, 2023-05-17:
very glad to hear the issue is resolved :)

closed 2023-05-17 by MrChromebox