MrChromebox/firmware #464: Unable to detect Intel i225 on Asus CN65

tirsojrp, 2023-05-07:
Installed an i225 2.5g to my Asus CN65 box using an A/E m2 to M key adapter. Tested in several minipcs (non coreboot):  card > Adapter > m2 wifi port. In this box the card is powered and activity leds works when I plug a cable. OS is pFsense.

MrChromebox, 2023-05-07:
cbmem log?

https://mrchromebox.tech/files/util/cbmem.tar.gz
unzip, run as `sudo ./cbmem -1 > cbmem.log`
probably need to run from Linux

tirsojrp, 2023-05-08:
[cbmem.log.gz](https://github.com/MrChromebox/firmware/files/11416334/cbmem.log.gz)

MrChromebox, 2023-05-08:
not sure what you did, but that's not a text log file

tirsojrp, 2023-05-08:
> not sure what you did, but that's not a text log file

[cbmem.log](https://github.com/MrChromebox/firmware/files/11417243/cbmem.log)

MrChromebox, 2023-05-08:
much better.

when coreboot scans the PCIe bus, it's not finding anything attached to the root port normally used for the WiFi (1c.3).

[INFO ]  Found PCIe Root Port #4 at PCI: 00:1c.3.
[DEBUG]  PCI: 00:1c.3 [8086/9d13] enabled
[DEBUG]  PCI: 00:1c.3 scanning...
[DEBUG]  PCI: pci_scan_bus for bus 02
[INFO ]  PCI: Static device PCI: 02:00.0 not found, disabling it.
[WARN ]  PCI: Leftover static devices:
[WARN ]  PCI: 02:00.0
[WARN ]  PCI: Check your devicetree.cb.
[DEBUG]  scan_bus: bus PCI: 00:1c.3 finished in 0 msecs

not sure what the issue could be offhand

tirsojrp, 2023-05-08:
I will try with the NVME port later and report back.

tirsojrp, 2023-05-13:
[cbmem_nvme.log](https://github.com/MrChromebox/firmware/files/11470533/cbmem_nvme.log)

No luck.

MrChromebox, 2023-05-13:
> [cbmem_nvme.log](https://github.com/MrChromebox/firmware/files/11470533/cbmem_nvme.log)
> No luck.

could have told you not to waste your time, that port is only wired for SATA, not PCIe

tirsojrp, 2023-06-15:
> > [cbmem_nvme.log](https://github.com/MrChromebox/firmware/files/11470533/cbmem_nvme.log)
> > No luck.
> 
> could have told you not to waste your time, that port is only wired for SATA, not PCIe

Actually, I put an NVME drive and it is detected.

MrChromebox, 2023-06-15:
> Actually, I put an NVME drive and it is detected.

sorry, got the Chromebox models mixed up, the CN60/62 do not support NVMe on the m.2 socket

I'm not sure why coreboot isn't detecting the device, the log isn't very helpful here

closed 2026-11-04 by MrChromebox