MrChromebox/firmware #466: [Stout] Thinkpad x131e thinks it's suspended on reboot (no fan)

cablespaghetti, 2023-05-14:
Hi,

I've got a couple of these devices and have run various versions of your firmware and various linux distros since early last year. First of all, thanks for the work put into making this possible!

The issue I've got is that when powered off and on, or suspended the devices work as normal. However when rebooted the little red indicator light blinks like the device is suspended and critically the fan does not cut in causing overheating issues. The workaround is to never reboot, only suspend or power off/on.

However I've recently put one into service as a laptop for my son, so it's being used more heavily and made me more keen to figure out how to fix this.

I'm pretty technical but know almost nothing about firmware. Let me know any I can do to help you debug this or figure it out myself.

Thanks again!

I believe these might be related to the same issue but I'm not certain. Also I'm not sure anyone has noticed the light blinking like the device is suspended:
https://github.com/MrChromebox/firmware/issues/131
https://www.reddit.com/r/chrultrabook/comments/b4ouqn/lenovo_x131e_chromebook_stout_fan_issues/

MrChromebox, 2023-05-14:
my x131e is no longer functional so I have no way to test this device currently. I'm not even sure what would cause these issues for this device only, since the other Sandybridge/Ivybridge platform devices do not have similar issues.

cablespaghetti, 2023-05-14:
Fair enough. It is super weird! I think it’s time I learn to debug firmware.

MrChromebox, 2023-05-14:
the problem with STOUT is that it's using a closed-source, non-Google EC. That makes it very hard to debug things like the fan control as there is limited info available about the EC-AP interface

cablespaghetti, 2023-05-14:
> the problem with STOUT is that it's using a closed-source, non-Google EC. That makes it very hard to debug things like the fan control as there is limited info available about the EC-AP interface

Thanks 2013 era Lenovo engineers! I'm convinced now from the blinky red light that whatever the firmware is doing for a reboot is making the EC think the device is in sleep. I suppose the debug process is going to be sending random signals to the EC and hoping for the best 😆

cablespaghetti, 2023-05-15:
I thought I'd experiment and install a SeaBIOS ROM from here: https://johnlewis.ie/custom-chromebook-firmware/rom-download/

The problem does not exist for that ROM. I would obviously much prefer to have a UEFI ROM which somebody is supporting, but it is interesting. Now I get to figure out what they do differently on reboot.

closed 2026-11-04 by MrChromebox