MrChromebox/firmware #468: lenovo 300e 2nd gen AST AMD

lx00xl, 2023-05-17:
i need lenovo 300e 2nd gen AST AMD or treeya360 stock firmware
i forgot to backup when flashing coreboot firmware
i can't find anywhere 
on Firmware Utility Script it says not found too

MrChromebox, 2023-05-17:
I don't have / don't provide firmware backups, keeping an updated inventory of all boards is just too much work.

You can use the crosfirmware script from the coreboot repo here: 
https://github.com/coreboot/coreboot/tree/master/util/chromeos

download and run: `bash crosfirmware.sh treeya360` (or maybe just `treeya`)

lx00xl, 2023-05-18:
I have try it
its downloading recovery firmware 
i mean i need the stock bios firmware
on your script its should be shellball.(codename of device).bin
i have try treeya and treeya360 but its not found
But i cant find it anywhere
or you have any suggestion where should i try to find it

MrChromebox, 2023-05-18:
I do not offer shellball images anymore. The firmware image extracted from the recovery image is the same thing.

running `bash crosfirmware.sh treeya360` absolutely works, I just did it here myself. It will produce the file coreboot-Google_Grunt.11031.182.0.bin which you can restore to your device

LerbagMonsa588, 2025-12-11:
i didnt use the instructions, but instead searched how to enter the boot menu (whatever the technicism is) with the escape, reload, and power buttons from which I selected a usb drive to install lubuntu, cheers