MrChromebox/firmware #469: secure boot doesn't work on doesn't work on Pixelbook Eve

jmontleon, 2023-05-21:
I saw in the note on the coreboot 4.20 release `Added UEFI SecureBoot support`.

On a 2017 Pixelbook (Eve), when going into the Secure Boot Configuration, Current Secure Boot Configuration is blank anddoesn't say Enabled or Disabled. I am also unable to enroll a PK key and I get a message,

```
ERROR: Enrollment failed!
        Error status: 2.
```

I walked through the process of generating a PK key, converting it to DER format, and enrolling it on a VM with EDK2 and was able to get secure boot working, so I think I'm doing it correctly.

Should this be possible now?

MrChromebox, 2023-05-21:
it seems that on some devices, you need to clear the NVRAM via the script after the update before secureboot will function properly - try that and LMK if that works

jmontleon, 2023-05-21:
That did it! Cleared, Rebooted, Enabled Secure Boot and worked perfectly.

Thank you!

closed 2023-06-10 by MrChromebox