MrChromebox/firmware #470: Taniks EMMC not seen in coreboot

Jay-Beanz, 2023-05-21:
Coreboot cannot find any UEFI partitions on my EMMC drive.

Also NVME drives dont seem to work, unseen in lspci, but unsure if this is a firmware or motherboard limitation.

Jay-Beanz, 2023-05-22:
There is two models of taniks, the i5 has a 256 nvme and the i3 has Emmc, so I assumed maybe only one was tested  I'll do a bit more testing when I get home and try to get you a log.

MrChromebox, 2023-05-22:
I'm not sure which was tested TBH. Post a cbmem log so I can see what's going on with coreboot

MrChromebox, 2023-05-22:
so, the logs indicate that your device shipped with eMMC, not NVMe. Unfortunately, just popping in a NVMe drive won't work, since the FW_CONFIG (eeprom attached to EC) tells coreboot your SKU has eMMC so it gets configured accordingly. We can work around that in coreboot though. I'm not sure why the eMMC isn't available in edk2 for booting though - I'd likely need serial output (via SuzyQ) and a debug-enabled build to see what's going on there

Jay-Beanz, 2023-05-22:
Alright, well let me know how i can help, still have my suzyQ setup from disabling WP and willing to get you any logs or info you'd need.

MrChromebox, 2023-05-22:
are you able to compile coreboot/edk2 from source?

Jay-Beanz, 2023-05-22:
Yeah i think so, i built your 4.20 UEFI image one day before you released lol, not sure if there's more to it though.

MrChromebox, 2023-05-22:
sync up the 4.10 branch and make the following edits:
- to configs/adl/config.taniks.uefi, 
remove the line:
`# CONFIG_CONSOLE_SERIAL is not set`
and add the following line:
`CONFIG_EDK2_DEBUG=y`
- to src/mainboard/google/brya/variants/taniks/overridetree.cb, remove the following lines:
`probe BOOT_NVME_MASK BOOT_NVME_ENABLED`
`probe BOOT_EMMC_MASK BOOT_EMMC_ENABLED`

save, then `./build-uefi.sh taniks` and flash. With the SuzyQ cable connected to another device running linux, use minicom (or another terminal program) to connect to /dev/ttyUSB1 and log a full boot for both the NVMe and eMMC

Jay-Beanz, 2023-05-22:
Ill give that a go and report back, I assume you mean the 4.20 branch?

Jay-Beanz, 2023-05-22:
[taniks-emmc.log](https://github.com/MrChromebox/firmware/files/11537749/taniks-emmc.log)
[taniks-nvme.log](https://github.com/MrChromebox/firmware/files/11537750/taniks-nvme.log)
With that change the NVME drive shows up in edk2, and in an os.
The emmc still doesn't appear in edk2

Jay-Beanz, 2023-05-23:
Yeah worked fine under Linux, was able to format and install. But never showed up on edk2 boot menu or was bootable. Didn't show up under windows installer but I assumed that was driver related. 

Up to you if you want to spend more time on it, I'm happy now that NVME is working.

MrChromebox, 2023-05-23:
unfortunately I'd probably need to get my hands on a device in order to adequately debug the issue, unless there's something I'm missing in the log

Jay-Beanz, 2023-05-23:
Yeah i understand, wish i could be of more help. I guess i just rebuild without debugging for now, should i re add: `probe BOOT_EMMC_MASK BOOT_EMMC_ENABLED` or would it be ` probe BOOT_NVME_MASK BOOT_NVME_ENABLED` to overridetree.cb?

MrChromebox, 2023-05-23:
I wouldn't re-add either, there's really no need. just disable the serial output

Jay-Beanz, 2023-05-23:
Sounds good! Thanks for the help, glad i was able to get it bootable now.

closed 2023-05-23 by Jay-Beanz

MrChromebox, 2023-05-23:
I'll go ahead and update the TANIKS release image with NVMe support for eMMC boards tho

MrChromebox, 2023-05-23:
release image updated :)