MrChromebox/firmware #471: Lenovo flex 5i (lillipup) oled brightness adjustment in windows is not functional

junnikokuki, 2023-05-23:
The brightness adjustment slider can slide and the hotkey can show, but the brightness remains unchanged. It works in ubuntu. I have another same model with lcd works as expected. Both laptops have same version coreboot and windows. (Just log here to wait if oled vbt updated.)

MrChromebox, 2023-05-23:
I'll need some info from Linux, `dmesg | grep drm > dmesg.log` when booted with 'drm.debug=0xe' added to kernel boot params, so I can see what the correct backlight control type and params are, in order to correct the VBT for Windows

junnikokuki, 2023-05-24:
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.19.0-41-generic root=UUID=20411209-c757-458d-9a85-415fe05c1a38 ro quiet splash vt.handoff=7 drm.debug=0xe
[    0.044764] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.19.0-41-generic root=UUID=20411209-c757-458d-9a85-415fe05c1a38 ro quiet splash vt.handoff=7 drm.debug=0xe
[    2.379106] systemd[1]: Starting Load Kernel Module drm...
[    2.398454] ACPI: bus type drm_connector registered
[    2.399045] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    2.399222] systemd[1]: Finished Load Kernel Module drm.
[    3.742102] i915 0000:00:02.0: [drm:intel_wopcm_init_early [i915]] WOPCM: 2048K
[    3.742220] i915 0000:00:02.0: [drm:__confirm_options [i915]] enable_guc=0 (guc:no submission:no huc:no slpc:no)
[    3.742315] i915 0000:00:02.0: [drm:intel_pch_type [i915]] Found Tiger Lake LP PCH
[    3.742391] i915 0000:00:02.0: [drm:get_allowed_dc_mask [i915]] Allowed DC state mask 4000000a
[    3.742535] i915 0000:00:02.0: [drm:intel_uncore_init_mmio [i915]] unclaimed mmio detected on uncore init, clearing
[    3.742650] i915 0000:00:02.0: [drm:intel_device_info_runtime_init [i915]] rawclk rate: 19200 kHz
[    3.742825] i915 0000:00:02.0: [drm:init_engine_mask [i915]] vdbox enable: 0005, instances: 0005
[    3.742882] i915 0000:00:02.0: [drm:init_engine_mask [i915]] vebox enable: 0001, instances: 0001
[    3.743067] i915 0000:00:02.0: [drm:ggtt_probe_hw [i915]] GGTT size = 4096M
[    3.743123] i915 0000:00:02.0: [drm:ggtt_probe_hw [i915]] GMADR size = 256M
[    3.743171] i915 0000:00:02.0: [drm:ggtt_probe_hw [i915]] DSM size = 60M
[    3.743428] i915 0000:00:02.0: [drm:i915_gem_init_stolen [i915]] GEN6_STOLEN_RESERVED = 0x000000007fa000c7
[    3.743499] i915 0000:00:02.0: [drm:i915_gem_init_stolen [i915]] Memory reserved for graphics device: 61440K, usable: 59392K
[    3.743611] i915 0000:00:02.0: [drm:intel_opregion_setup [i915]] graphic opregion physical addr: 0x76b42000
[    3.743692] i915 0000:00:02.0: [drm:intel_opregion_setup [i915]] ACPI OpRegion version 2.0.0
[    3.743772] i915 0000:00:02.0: [drm:intel_opregion_setup [i915]] Public ACPI methods supported
[    3.743844] i915 0000:00:02.0: [drm:intel_opregion_setup [i915]] ASLE supported
[    3.743902] i915 0000:00:02.0: [drm:intel_opregion_setup [i915]] ASLE extension supported
[    3.743967] i915 0000:00:02.0: [drm:intel_opregion_setup [i915]] Found valid VBT in ACPI OpRegion (RVDA)
[    3.744030] i915 0000:00:02.0: [drm:intel_dram_detect [i915]] DRAM channels: 4
[    3.744090] i915 0000:00:02.0: [drm:intel_dram_detect [i915]] Watermark level 0 adjustment needed: no
[    3.744146] i915 0000:00:02.0: [drm:icl_get_qgv_points.constprop.0 [i915]] QGV 0: DCLK=2134 tRP=20 tRDPRE=16 tRAS=45 tRCD=20 tRC=65
[    3.744223] i915 0000:00:02.0: [drm:icl_get_qgv_points.constprop.0 [i915]] QGV 1: DCLK=3201 tRP=29 tRDPRE=20 tRAS=68 tRCD=29 tRC=97
[    3.744294] i915 0000:00:02.0: [drm:icl_get_qgv_points.constprop.0 [i915]] QGV 2: DCLK=4268 tRP=39 tRDPRE=24 tRAS=90 tRCD=39 tRC=129
[    3.744359] i915 0000:00:02.0: [drm:icl_get_qgv_points.constprop.0 [i915]] QGV 3: DCLK=2668 tRP=24 tRDPRE=18 tRAS=56 tRCD=24 tRC=80
[    3.744418] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW0 / QGV 0: num_planes=0 deratedbw=18910
[    3.744475] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW0 / QGV 1: num_planes=0 deratedbw=23413
[    3.744531] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW0 / QGV 2: num_planes=0 deratedbw=26223
[    3.744592] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW0 / QGV 3: num_planes=0 deratedbw=21569
[    3.744651] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW1 / QGV 0: num_planes=4 deratedbw=23413
[    3.744706] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW1 / QGV 1: num_planes=4 deratedbw=31053
[    3.744760] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW1 / QGV 2: num_planes=4 deratedbw=34000
[    3.744813] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW1 / QGV 3: num_planes=4 deratedbw=27627
[    3.744865] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW2 / QGV 0: num_planes=0 deratedbw=26577
[    3.744915] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW2 / QGV 1: num_planes=0 deratedbw=34000
[    3.744968] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW2 / QGV 2: num_planes=0 deratedbw=34000
[    3.745027] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW2 / QGV 3: num_planes=0 deratedbw=32141
[    3.745085] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW3 / QGV 0: num_planes=0 deratedbw=28503
[    3.745139] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW3 / QGV 1: num_planes=0 deratedbw=34000
[    3.745192] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW3 / QGV 2: num_planes=0 deratedbw=34000
[    3.745243] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW3 / QGV 3: num_planes=0 deratedbw=34000
[    3.745294] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW4 / QGV 0: num_planes=0 deratedbw=29574
[    3.745350] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW4 / QGV 1: num_planes=0 deratedbw=34000
[    3.745407] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW4 / QGV 2: num_planes=0 deratedbw=34000
[    3.745461] i915 0000:00:02.0: [drm:tgl_get_bw_info.isra.0 [i915]] BW4 / QGV 3: num_planes=0 deratedbw=34000
[... cut ...]
[  120.144931] [drm:drm_mode_addfb2 [drm]] [FB:370]
[  120.561891] [drm:drm_mode_addfb2 [drm]] [FB:372]
[  120.585543] [drm:drm_mode_addfb2 [drm]] [FB:374]
[  120.868067] [drm:drm_mode_addfb2 [drm]] [FB:372]
[  120.892966] [drm:drm_mode_addfb2 [drm]] [FB:370]
[  120.944910] [drm:drm_mode_addfb2 [drm]] [FB:372]
[  120.962787] [drm:drm_mode_addfb2 [drm]] [FB:370]
[  121.534353] [drm:drm_mode_addfb2 [drm]] [FB:372]
[  122.136044] [drm:drm_mode_addfb2 [drm]] [FB:370]
[  122.271844] [drm:drm_mode_addfb2 [drm]] [FB:372]

closed 2026-11-04 by MrChromebox