MrChromebox/firmware #472: Taeko4es with coreboot 4.20 won't boot. Can I debug it?

junnikokuki, 2023-05-23:
Hi,
    I have a taeko4es with i3 0000, it has a working cros. I built coreboot 4.20 with TAEKO4ES config and the 3rdparty blob is extracted from the shipped cros firmware(I backup it with suzyq) and ec.RW.flat from taeko folder. But it can not boot. The power light is on for a second and then off then on and the screen no light. Is it a problem of config or spd? Can I debug it with the suzyq cable?

junnikokuki, 2023-05-29:
[https://gist.github.com/junnikokuki/ae54c84916810daa58ade4ab9dba7568](url)

junnikokuki, 2023-05-29:
seems a boot loop, no idea where the problem is.

MrChromebox, 2023-05-29:
looks to me like cse_fw_sync() is calling for a reset after executing, but not sure why since it's already in RW mode and up to date.

junnikokuki, 2023-05-30:
It's an ec update progress? I just not using the ec from original firmware, but that in production taeko folder(ec.RW.flat).

junnikokuki, 2023-07-19:
Managed to compile coreboot for the taeko4es and installed win10. It works but when I use the touchpad, the cpu usage will be more than 30%. The cpu will keep up than 30% even when I stop touch the pad. It will be back to normal when I plug or unplug the ac power. Keyboard, touch screen and external mouse work ok. 

I asked on discord and coolstar said "GPIO IRQ problem in coreboot, some GPIO is locked clearly"

How can I fix it?

junnikokuki, 2023-07-19:
I have ever installed ubuntu and it's idle cpu usage is also more than 20%. So I think it exist. If it helps, I can reinstall a ubuntu.

junnikokuki, 2023-07-22:
> 

I think the gpio is touchpad related. If I was in ubuntu, how can I detect the bug?

MrChromebox, 2023-07-22:
IDK, maybe look at /proc/interrupts ?

github isn't the best place to try and debug this kind of issue

coolstar, 2023-07-23:
This was the fix for the issue on banshee (which was very similar to the issue described here): https://github.com/MrChromebox/coreboot/commit/f564c8b34abbc7010871cfa426af7ee7261c3fde

junnikokuki, 2023-07-24:
> This was the fix for the issue on banshee (which was very similar to the issue described here): [MrChromebox/coreboot@f564c8b](https://github.com/MrChromebox/coreboot/commit/f564c8b34abbc7010871cfa426af7ee7261c3fde)

OK, I will try it. Thanks.

MrChromebox, 2023-07-24:
@junnikokuki there's nothing to try, the fix for banshee applies to all brya baseboard boards, taeko4es included. It's already done

junnikokuki, 2023-11-05:
> This was the fix for the issue on banshee (which was very similar to the issue described here): [MrChromebox/coreboot@f564c8b](https://github.com/MrChromebox/coreboot/commit/f564c8b34abbc7010871cfa426af7ee7261c3fde)

I tried 4.19, 4.20 and 4.21, all the es alder lake chromebooks I tested(anahera4es, redrix4ex, primus4es and taeko4es) had the problem that the first p-core's first thread always has high cpu usage, whenever busy or idle. I use performace analyzer to debug and it pointed to system interrupt or acpi.sys(acpireadgpestatusregister, halpacpipmregisterreadport). But the release version alder lake cpu's chromebooks do not have this problem. So how can I debug it?

MrChromebox, 2023-11-05:
are these all pre-production devices using engineering sample (ES) CPUs? If so, then there's no telling what the issue is

junnikokuki, 2023-11-06:
Yes they are. But the ubuntu acts well, chromeos too.

junnikokuki, 2023-11-07:
@coolstar Any suggestion?

closed 2026-11-04 by MrChromebox