MrChromebox/firmware #478: Chromebook freezes when selecting enroll from disk in MOK Manager

Mukul1127, 2023-06-12:
When selecting Enroll Key or Enroll Hash, the chromebook just freezes up until I reset it. Installing though Device Manager DB Options works though.
[image]

Mukul1127, 2023-06-14:
MOK Manager comes with Ventoy but that doesn’t explain why it freezes.

MrChromebox, 2023-06-14:
no idea, never used it

MrChromebox, 2023-06-14:
the only way to debug edk2 is with a serial/debug-enabled build and serial output from the device being used. That's not really practical to do on GLIMMER without some soldering and a custom debug cable

Mukul1127, 2023-06-16:
Don’t have a soldering iron and don’t plan too. Closing this issue

closed 2023-06-16 by Mukul1127

Mukul1127, 2023-11-22:
This randomly started working again, probably because of a firmware update.