MrChromebox/firmware #482: Wake on WLAN (WoWLAN) support?

bam80, 2023-06-17:
I didn't find any information if the WoWLAN feature is supported, particularly on Chromebox 3.
Could you help please?

Kernel doc:
https://wireless.wiki.kernel.org/en/users/documentation/wowlan

bam80, 2023-06-23:
@MrChromebox any hope?
I tried to setup my system to wake on magic backed but had no luck so far - I see WiFi connection persists even during suspend but I can remotely wake the system up by sending magic packet to it..

bam80, 2023-06-23:
Thank you very much, it's quite important for remote access to my system.

bam80, 2023-06-25:
Here it's listed what is needed for WoLAN (and I assume WoWLAN too) support on coreboot, but that post is 9 yeas old:
https://coreboot.coreboot.narkive.com/fetipAgr/wol-pci-pme-wakeup-from-s3-baytrail-soc-bayleybay-crb#post4

bam80, 2023-06-29:
> But I'll look into it a bit more

Will it be possible to look into it in the foreseeable future?

MrChromebox, 2023-06-29:
I'm not sure when I'll have time to do so unfortunately

closed 2026-11-04 by MrChromebox