MrChromebox/firmware #484: chromebook cant connect/detect google on first setup after firmware recovery

lx00xl, 2023-06-18:
recently, i change back from coreboot bios to original bios (Chrome OS)
on the first setup after firmware recovery, i try several wifi but it says still cant connect to Google
when changing to original via mrchromebox firmware utility, i try change the HWID
Maybe it is the cause
any solution for my problem?

MrChromebox, 2023-06-18:
sounds like the date/time are too far off for it to sync, or time server not accessible. 

Why would you think it would be a  good idea to change the HWID? That will only cause problems

lx00xl, 2023-06-19:
i am following wrong suggestion. 
is there any pattern for HWID?
i forgot to backup mine

MrChromebox, 2023-06-19:
what is your device/board name? what is the HWID currently set to?

lx00xl, 2023-06-20:
grunt/treeya360
im following the example on your firmware utiliy, so i write TREEYA360 TEST 3456 (4 random number)
i read, maybe on reddit, some people doing it and its fine

MrChromebox, 2023-06-20:
@lx00xl the 'TEST' string is invalid, it's the default when no HWID is set. Since you messed it up, you need to set it back to a valid one. Try: TREEYA360 ABC-123-DEF-456

closed 2026-11-04 by MrChromebox