MrChromebox/firmware #486: Unable to unlock CFGLock

wenwutang, 2023-06-21:
I want to install macos on hp elite c1030 chromebook JinLon (i7 10610u), but can't unlock CFGLock of Bios.

MrChromebox, 2023-06-22:
I don't understand exactly what you're asking / what the problem is

wenwutang, 2023-06-22:
Sorry, my English is not good, I use translation software.I want to install hackintosh on hp elite c1030 chromebook。But it is always unsuccessful and stuck at [EB|LOG:EXITBS:START], I guess it should be related to CFGLock。Unable to unlock using traditional bios unlocking method.In line 3120 of the source file 'src/vendorcode/intel
/edk2/edk2-stable202005/MdePkg/Include/Register/Intel/MsrSkylakeMsr.h', there is a variable CFGLock whose value is 1

wenwutang, 2023-06-22:
I don't know if my guess is correct

MrChromebox, 2023-06-22:
please check on the Hackintosh channel of the chrultrabook Discord server, I'm pretty sure others there can help you work around the issue without any firmware changes

closed 2026-11-04 by MrChromebox