MrChromebox/firmware #488: Chromebook 314 CB314-3HT bios write protection screw

Maximilian2808, 2023-06-22:
Hi Everyone, I have a Chromebook 314 CB314-3HT with an Intel Pentium Silver N6000 processor and I wanted to install a different operating system on the laptop called "Tails". However, I keep getting an error message when I want to use the ChromeOS firmware utility script, it comes the following error message, "hardware write-protect enabled cannot flash full rom firmware" can someone help me?

I thank you in advance!

Translated with www.DeepL.com/Translator (free version)

MrChromebox, 2023-06-22:
the type of write protection used for each device is listed on https://mrchromebox.tech/#devices. if more info is available for that device, there is a link  as well

Maximilian2808, 2023-06-22:
> the type of write protection used for each device is listed on https://mrchromebox.tech/#devices. if more info is available for that device, there is a link as well

Here is more information
[image]

MrChromebox, 2023-06-22:
did you read the link I provided? none of the info in that pic matters. Only the ChromeOS board name / HWID.

that said, the CB314-3HT is listed on the page. That's all the info I have on the matter

Maximilian2808, 2023-06-22:
> did you read the link I provided? none of the info in that pic matters. Only the ChromeOS board name / HWID.
> 
> that said, the CB314-3HT is listed on the page. That's all the info I have on the matter

So on the list I find the Chromebook 314 CB314, and then behind it is the Cr50 method, I have already tried the method and several times, but I had no success with it!

MrChromebox, 2023-06-22:
> So on the list I find the Chromebook 314 CB314, and then behind it is the Cr50 method, I have already tried the method and several times, but I had no success with it!

which part of 'model # is not deterministic' was unclear?

And what do you mean you tried the method? what did you do exactly? You need to use a SuzyQ cable and follow the linked instructions. or use a jumper.

Maximilian2808, 2023-06-22:
> which part of 'model # is not deterministic' was unclear?
>
> And what do you mean you tried the method? what did you do exactly? You need to use a SuzyQ cable and follow the linked instructions. or [...]

Okay thank you very much, I have also already received help on fem discord server. I will build my own SuzyQ and follow the steps.

closed 2023-07-20 by MrChromebox