MrChromebox/firmware #489: Firmware isn't initializing TPM PCRs

dankcatlord, 2023-06-22:
When I run tpm2_pcrread, PCRs 0-7 isnt being set by the firmware. I'm running ArchLinux on a Samsung Chromebook 4+ with the 4.20.0 firmware. This issue prevents me from sealing keys in my TPM securely. I think the firmware might not be initializing the TPM on startup.
```
# tpm2_pcrread
  sha1:
    0 : 0x0000000000000000000000000000000000000000
    1 : 0x0000000000000000000000000000000000000000
    2 : 0x0000000000000000000000000000000000000000
    3 : 0x0000000000000000000000000000000000000000
    4 : 0x0000000000000000000000000000000000000000
    5 : 0x0000000000000000000000000000000000000000
    6 : 0x0000000000000000000000000000000000000000
    7 : 0x0000000000000000000000000000000000000000
    8 : 0x0000000000000000000000000000000000000000
    9 : 0x0000000000000000000000000000000000000000
    10: 0x0000000000000000000000000000000000000000
    11: 0x0000000000000000000000000000000000000000
    12: 0x0000000000000000000000000000000000000000
    13: 0x0000000000000000000000000000000000000000
    14: 0x0000000000000000000000000000000000000000
    15: 0x0000000000000000000000000000000000000000
    16: 0x0000000000000000000000000000000000000000
    17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    23: 0x0000000000000000000000000000000000000000
  sha256: ... # Same result for the respective hash lengths
  sha384: ...
  sha512: ...
```

dankcatlord, 2023-06-22:
Ah ok, is measured boot planned to be enabled in the future?

MrChromebox, 2023-06-22:
wasn't planning on it, but I can do so on a test build and see if that resolves the issue for you. shoot me an email or msg on Discord

coolstar, 2023-06-24:
PCRs can’t be written by edk2 on Bluebird because there’s no CR50 driver

dankcatlord, 2023-06-24:
> PCRs can’t be written by edk2 on Bluebird because there’s no CR50 driver

The full uefi image test build with measured boot enabled fills in PCR 2 of the sha256 bank with data but none of the other PCRs. AFAIK from looking at the [MrChromebox/coreboot](https://github.com/MrChromebox/coreboot)  repo, there is some support for CR50 in version 4.20.0 looking at the commits.

closed 2026-11-04 by MrChromebox