MrChromebox/firmware #490: Cannot recover to ChromeOS from UEFI, Dell 3100 2 in 1, Grabbiter

Filip722, 2023-06-24:
Hi,

I've had my fun with UEFI and different distros, but i would like to return to ChromeOS.

The laptop is a Dell 3100 2 in 1, Grabbiter.
No matter what i do, i cannot get the ChromeOS recovery from an image to complete.

Firstly, i have ran the Firmware utility script (while on Linux Mint) to restore to stock firmware without my backup file (I had it, but wanted to check what would happen). All went fine, had the "FLEEX" device name in the boot process, my recovery image was recognized, but failed after a couple of minutes.

Secondly, I reflashed (With the CH341a) eeprom with my backup file (made with your tool) with your guide on the wiki. Now i saw the "GRABBITER" name in the boot process, but recovery failed same as before.

Then i again reflashed UEFI with SPI flasher, worked fine, booted linux, ran the Firmware utility script and selected my backup file, which also flashed fine, but had the same result as in step 2 (Name correct, recovery failed).

What's the problem ChromeOS recovery is having?
I have used different usb flash drives, made the recovery on windows and linux (via the dd command on the chromeos webpage)

[Here are my logs from the usb stick](https://pastebin.com/vEaQa3tt), but i don't know from which step are they as there were multiple tries.

Please help me get back to ChromeOS

MrChromebox, 2023-06-24:
perform a recovery with the internal battery disconnected

Filip722, 2023-06-25:
> perform a recovery with the internal battery disconnected

It worked. Thank you very much

closed 2023-06-25 by Filip722