MrChromebox/firmware #492: OS verification does not activate

IvanPedroMuricy, 2023-07-11:
I was trying to install linux on my chromebook; after some failed attempt, I did started the recovery procedures. However, after inserting the recovery media USB and waiting for the chromebook to complete the whole process, once the I removed the USB and the chromebook started, nothing happened. In other words, it is requested that the space bar is pressed, but if you do it, nothing happens; if you do not do it, nothing happens either. What could be happening ?

Thank you.

MrChromebox, 2023-07-11:
does anything show in the upper left when you press spacebar? 
How about when pressing tab on OS verification off screen?
can you boot ChromeOS via CTRL+D?

IvanPedroMuricy, 2023-07-11:
Thank you for answering. 
Initially, it beeped when space bar was pressed, then a text in white on black appeared saying that the action was not possible because there was an error which forbade the operation. The same text you have in your website. However, I am not being able to cut and paste to show you. Well, later on, when I pressed "escape+refresh+power", it would inform that OS verification was off, but nothing would happen once I pressed the space bar. I did even follow some advice and did not press spacebar for a long time; however, nothing happened.

MrChromebox, 2023-07-11:
< Initially, it beeped when space bar was pressed, then a text in white on black appeared saying that the action was not possible because there was an error which forbade the operation.

not an error, and intentional setting of the GBB Flags to ensure a successful boot when the stock firmware was restored.

Press CTRL+D on the developer mode boot screen (*not* from recovery mode)  to boot ChromeOS, then set the flags back to default using my script.

IvanPedroMuricy, 2023-07-11:
I did that, but, once the computer turns on it requests and SD or a USB for recovery. After inserting the USB, the recovery process starts, completes, the screen requesting to press the space bar appears, but nothing else happens. The picture is attached.
[image]

MrChromebox, 2023-07-11:
SMH. It's not *requesting that you press space, it's saying if you want to exit developer mode, press space to return to verified boot (normal) mode. But you can't exit developer mode if the GBB flags are set to force it on, so you have to boot ChromeOS and reset the flags.

IvanPedroMuricy, 2023-07-12:
Greetings,
Hope I am not pestering you way too much with my lack of knowledge. If I am, please forgive me. In fact, this is the second chromebook I am working with. The first one, in which I used the legacy boot mode, an old samsung 3, runs on cruchbang++. 

Considering what you wrote above:  "(...) you have to boot ChromeOS and reset the Flags." This is exactly my doubt: do I need to press any special key so ChromeOS will be booted ? I guess the recovery media should also be inserted. Am I wrong ? Once again, thank you for your help. I do not how whether you could suggest a website (yours or otherwise) and/or a tutorial on the process.

MrChromebox, 2023-07-12:
>  This is exactly my doubt: do I need to press any special key so ChromeOS will be booted ?

* remove all USB
* power on device
* wait for 'OS verification off' screen to show
* press CTRL+D to boot ChromeOS
* open terminal, run my script as per instructions on site
* set firmware boot flags (GBB Flags) to factory default
* reboot
* press space to exit developer mode when prompted (assuming that's what you want to do)

> I guess the recovery media should also be inserted.

no, absolutely not. You've performed a successful recovery, so the media is no longer needed

IvanPedroMuricy, 2023-07-13:
Good morning. 
Thanks a lot for your instructions. I did exactly as you suggested (not even the mouse was plugged in), however, no change happened. I suspect somthing during configuration might have affected keyboard functions because only the function keys are responsive. None of the others react. 
Any further suggestions ? If not, thank you for your kind assistance.

closed 2026-11-04 by MrChromebox