MrChromebox/firmware #493: No fan control on Asuka

Samfire82, 2023-07-13:
The fan on my Asuka isn't running in the os and uefi menu. I have tried the fan control utility but it doesn't pick up the sensors,
The fan only comes on when i do Refresh+Power and ESC+Refresh+Power. The fan doesn't run in the os or uefi options 
How do i get the fan working

MrChromebox, 2023-07-13:
why do you think it should be running?

Chromebooks with fans are controlled by the EC, and set to auto mode on boot. You can override it from Linux using Google's ectool but you likely need to compile that yourself

MrChromebox, 2023-07-14:
you can try using https://mrchromebox.tech/files/util/ectool.tar.gz

Samfire82, 2023-07-14:
I have tried the command but i'm getting
ioctl -1, errno 25 (Inappropriate ioctl for device), EC result 255 (<unknown>)
ioctl -1, errno 25 (Inappropriate ioctl for device), EC result 255 (<unknown>)
ioctl -1, errno 25 (Inappropriate ioctl for device), EC result 255 (<unknown>)
ioctl -1, errno 25 (Inappropriate ioctl for device), EC result 255 (<unknown>)
ioctl -1, errno 25 (Inappropriate ioctl for device), EC result 255 (<unknown>)

Am i doing something wrong?

MrChromebox, 2023-07-14:
sounds like that version of ectool doesn't support ASUKA or your kernel is preventing access. Not sure what to tell you. You can try cloning the 'firmware-glados' branch of my Chrome-EC repo and building ectool from there.

Are you experiencing overly high temps or CPU throttling? Or just paranoid because you can't hear the fan?

Samfire82, 2023-07-15:
I'm experiencing high temperatures while multitasking and general tasks, for me it's typing on word (i run linux now on a usb) and just general tasks
The temps can get uncomfortable at times a

closed 2026-11-04 by MrChromebox