MrChromebox/firmware #495: Acer chromebook 315 numpad, page up/down, delete, *,-, enter, /, + not working

tjd86, 2023-07-17:
numpad, page up/down, delete, *,-, enter, /, + not working on Acer Chromebook 315 N4020 processor 4GB Ram with MrChromebox UEFI BIOS. I have tried several linux distros and the same buttons do not work. I have also tried keymaps from Gallium OS in Ubuntu.

Thanks

tjd86, 2023-07-23:
Since this issue is being passed over without a response, I give MrChromebox and his team a F+, thank you for your response. You have been of no help. Stop posting that you support a model but have no intention of doing so.

MrChromebox, 2023-07-23:
> Since this issue is being passed over without a response, I give MrChromebox and his team a F+, thank you for your response. You have been of no help. Stop posting that you support a model but have no intention of doing so.

And I give you a F- for open source project etiquette, and general lack of politeness.

The firmware, scripts, and documentation I provide is all done free of charge, basically entirely by myself.  I get dozens of messages for support every day via email, twitter, reddit, discord, slack, IRC, github, etc. Not all of them get answered the same day. Some get missed entirely. 

life tip of the day: pouting about not getting free help for a free project never motivates people to do anything other than write you off as an ungrateful entitled asshole.

A simple comment bumping the issue would have pushed this back to the top of the email queue, or you could have reached out via another method.

coolstar, 2023-07-23:
@tjd86 you're on your own regarding keyboard support in Linux, since that depends on the distro that you're using. Not on the firmware

closed 2023-12-10 by tjd86