MrChromebox/firmware #496: First boot good, subsequent boots crash once, then boot slowly

CraigDC, 2023-07-18:
I have installed UEFI firmware and ChromeOS Flex on an HP, Dell and Asus Chromebook.
The Asus boots fine every time.
The Dell and HP Chromebooks boot fine the first time (before OS Flex installed), but subsequent boots crash once then boot slowly.
Below is an excerpt of the log file when I think the trouble begins.
Any assistance would be greatly appreciated!
Thanks!
Craig

2023-07-17T00:37:52.816232Z NOTICE temp_logger[4877]:  x86_pkg_temp:54C iwlwifi_1:45C PL1:9.999W
2023-07-17T00:37:53.748454Z ERR cryptohomed[797]: Failed to call method: org.chromium.ChromeFeaturesServiceInterface.IsFeatureEnabled: object_path= /org/chromium/ChromeFeaturesService: org.freedesktop.DBus.Error.InvalidArgs: Unexpected feature name.
2023-07-17T00:37:53.748847Z WARNING cryptohomed[797]: No valid keysets on disk for (HASH:0022 1)
2023-07-17T00:37:53.760613Z INFO cryptohomed[797]: Adding random reset secret for UserSecretStash.
2023-07-17T00:37:53.763063Z WARNING cryptohomed[797]: Cryptohome Error reported on DBus API: Loc: kLocAuthSessionInvalidBlockTypeInAddAuthFactor/614: Loc: kLocAuthBlockUtilNoSupportedInGetAuthBlockWithType/2717: Loc: kLocPinWeaverAuthBlockHwsecNotReadyInIsSupported/2723
2023-07-17T00:37:53.763220Z INFO cryptohomed[797]: Reporting cryptohome error hashed stack 2186136912 from 614-2717-2723
2023-07-17T00:37:53.818877Z WARNING crash_reporter[4879]: Received crash notification for jserror[2155] user 1000 (called directly)
2023-07-17T00:37:53.823782Z INFO crash_reporter[4879]: Accessing crash dir '/run/daemon-store/crash/(HASH:0022 1)' via symlinked handle '/proc/self/fd/10'
2023-07-17T00:37:53.832225Z WARNING crash_reporter[4880]: Received crash notification for jserror[2155] user 1000 (called directly)
2023-07-17T00:37:53.835069Z INFO crash_reporter[4880]: Accessing crash dir '/run/daemon-store/crash/(HASH:0022 1)' via symlinked handle '/proc/self/fd/10'
2023-07-17T00:37:53.870142Z INFO crash_reporter[4879]: Finishing crash. Meta file: /proc/self/fd/10/jserror.20230717.003753.59458.2155.meta
2023-07-17T00:37:53.872873Z INFO crash_reporter[4880]: Finishing crash. Meta file: /proc/self/fd/10/jserror.20230717.003753.76763.2155.meta
2023-07-17T00:37:55.206421Z WARNING session_manager[1816]: WARNING session_manager: [device_identifier_generator.cc(228)] No device identifiers available, no state keys generated
2023-07-17T00:38:08.364829Z WARNING dlcservice[2790]: WARNING dlcservice: [dlc.cc(309)] Scaled DLC=assistant-dlc will not update with the OS.
2023-07-17T00:38:08.367577Z WARNING dlcservice[2790]: WARNING dlcservice: [dlc.cc(309)] Scaled DLC=nc-ap-dlc will not update with the OS.
2023-07-17T00:38:08.658063Z NOTICE kernel: [  237.663448] Lockdown: flashrom: raw io port access is restricted; see man kernel_lockdown.7
2023-07-17T00:38:08.662219Z NOTICE kernel: [  237.667316] Lockdown: flashrom: raw io port access is restricted; see man kernel_lockdown.7
2023-07-17T00:38:13.201032Z NOTICE kernel: [  242.204702] Lockdown: flashrom: raw io port access is restricted; see man kernel_lockdown.7
2023-07-17T00:38:13.204017Z NOTICE kernel: [  242.207511] Lockdown: flashrom: raw io port access is restricted; see man kernel_lockdown.7
2023-07-17T00:38:13.210028Z NOTICE kernel: [  242.213237] Lockdown: flashrom: raw io port access is restricted; see man kernel_lockdown.7
2023-07-17T00:38:13.212017Z NOTICE kernel: [  242.215952] Lockdown: flashrom: raw io port access is restricted; see man kernel_lockdown.7
2023-07-17T00:38:13.213743Z NOTICE activate_date[4935]: activate_date failed.
2023-07-17T00:38:20.460123Z ERR cryptohomed[797]: Failed to call method: org.chromium.ChromeFeaturesServiceInterface.IsFeatureEnabled: object_path= /org/chromium/ChromeFeaturesService: org.freedesktop.DBus.Error.InvalidArgs: Unexpected feature name.
2023-07-17T00:38:20.460462Z WARNING cryptohomed[797]: No valid keysets on disk for (HASH:0022 1)
2023-07-17T00:38:20.469230Z INFO cryptohomed[797]: Adding random reset secret for UserSecretStash.
2023-07-17T00:38:20.470783Z WARNING cryptohomed[797]: Cryptohome Error reported on DBus API: Loc: kLocAuthSessionInvalidBlockTypeInAddAuthFactor/614: Loc: kLocAuthBlockUtilNoSupportedInGetAuthBlockWithType/2717: Loc: kLocPinWeaverAuthBlockHwsecNotReadyInIsSupported/2723
2023-07-17T00:38:20.470957Z INFO cryptohomed[797]: Reporting cryptohome error hashed stack 2186136912 from 614-2717-2723
2023-07-17T00:38:38.444225Z ERR cryptohomed[797]: Failed to call method: org.chromium.ChromeFeaturesServiceInterface.IsFeatureEnabled: object_path= /org/chromium/ChromeFeaturesService: org.freedesktop.DBus.Error.InvalidArgs: Unexpected feature name.
2023-07-17T00:38:38.444546Z WARNING cryptohomed[797]: No valid keysets on disk for (HASH:0022 1)
2023-07-17T00:38:38.453940Z INFO cryptohomed[797]: Adding random reset secret for UserSecretStash.
2023-07-17T00:38:38.455705Z WARNING cryptohomed[797]: Cryptohome Error reported on DBus API: Loc: kLocAuthSessionInvalidBlockTypeInAddAuthFactor/614: Loc: kLocAuthBlockUtilNoSupportedInGetAuthBlockWithType/2717: Loc: kLocPinWeaverAuthBlockHwsecNotReadyInIsSupported/2723
2023-07-17T00:38:38.456085Z INFO cryptohomed[797]: Reporting cryptohome error hashed stack 2186136912 from 614-2717-2723
2023-07-17T00:38:52.836020Z NOTICE temp_logger[4968]:  x86_pkg_temp:53C iwlwifi_1:43C PL1:9.999W
2023-07-17T00:38:55.207789Z WARNING session_manager[1816]: WARNING session_manager: [device_identifier_generator.cc(228)] No device identifiers available, no state keys generated
2023-07-17T00:38:55.636928Z ERR cryptohomed[797]: Failed to call method: org.chromium.ChromeFeaturesServiceInterface.IsFeatureEnabled: object_path= /org/chromium/ChromeFeaturesService: org.freedesktop.DBus.Error.InvalidArgs: Unexpected feature name.
2023-07-17T00:38:55.637345Z WARNING cryptohomed[797]: No valid keysets on disk for (HASH:0022 1)
2023-07-17T00:38:55.648485Z INFO cryptohomed[797]: Adding random reset secret for UserSecretStash.
2023-07-17T00:38:55.650445Z WARNING cryptohomed[797]: Cryptohome Error reported on DBus API: Loc: kLocAuthSessionInvalidBlockTypeInAddAuthFactor/614: Loc: kLocAuthBlockUtilNoSupportedInGetAuthBlockWithType/2717: Loc: kLocPinWeaverAuthBlockHwsecNotReadyInIsSupported/2723
2023-07-17T00:38:55.650870Z INFO cryptohomed[797]: Reporting cryptohome error hashed stack 2186136912 from 614-2717-2723
2023-07-17T00:39:35.083452Z INFO iioservice[568]: INFO iioservice: [iio_sensor.cc(56)] (2) Request(): Received SensorService from Mojo Service Manager
2023-07-17T00:39:35.343578Z WARNING chrome[2155]: 2023-07-17T00:39:35.343457Z WARNING chrome[2155:2155]: [chrome_content_verifier_delegate.cc(215)] Corruption detected in extension jkghodnilhceideoidjikpgommlajknk installed at: /usr/share/chromeos-assets/input_methods/input_tools, from webstore: 0, corruption reason: 1, should be repaired: 1, extension location: kComponent
2023-07-17T00:39:35.345822Z WARNING chrome[2155]: 2023-07-17T00:39:35.345727Z WARNING chrome[2155:2155]: [chrome_content_verifier_delegate.cc(215)] Corruption detected in extension jkghodnilhceideoidjikpgommlajknk installed at: /usr/share/chromeos-assets/input_methods/input_tools, from webstore: 0, corruption reason: 1, should be repaired: 1, extension location: kComponent
2023-07-17T00:39:35.438949Z WARNING chrome[2155]: 2023-07-17T00:39:35.438847Z WARNING chrome[2155:2155]: [chrome_content_verifier_delegate.cc(215)] Corruption detected in extension jkghodnilhceideoidjikpgommlajknk installed at: /usr/share/chromeos-assets/input_methods/input_tools, from webstore: 0, corruption reason: 1, should be repaired: 1, extension location: kComponent
2023-07-17T00:39:35.627452Z WARNING chrome[2155]: 2023-07-17T00:39:35.627354Z WARNING chrome[2155:2155]: [chrome_content_verifier_delegate.cc(215)] Corruption detected in extension jkghodnilhceideoidjikpgommlajknk installed at: /usr/share/chromeos-assets/input_methods/input_tools, from webstore: 0, corruption reason: 1, should be repaired: 1, extension location: kComponent
2023-07-17T00:39:45.141919Z INFO iioservice[568]: INFO iioservice: [sensor_service_impl.cc(417)] (2) OnSensorServiceDisconnect(): SensorDevice disconnected. ReceiverId: 3
2023-07-17T00:39:50.002725Z ERR shill[649]: ERROR shill: [rtnl_message.cc(1103)] RTNL address message does not have a valid IFA_ADDRESS
2023-07-17T00:39:51.189914Z INFO periodic_scheduler[2724]: crash_sender: running /sbin/crash_sender
[... cut ...]
2023-07-17T00:47:40.809021Z INFO kernel: [  809.529296] perf: interrupt took too long (10588 > 10492), lowering kernel.perf_event_max_sample_rate to 18000
2023-07-17T00:47:41.454054Z NOTICE kernel: [  810.173541] Lockdown: cbmem: /dev/mem,kmem,port is restricted; see man kernel_lockdown.7
2023-07-17T00:47:42.060920Z INFO vm_cicerone[4133]: Received GetDebugInformation request
2023-07-17T00:47:42.502032Z NOTICE kernel: [  811.221313] Lockdown: xz: debugfs access is restricted; see man kernel_lockdown.7
2023-07-17T00:47:42.811555Z INFO cups_launcher[1833]: Launching cupsd with config /etc/cups/cupsd.conf
---------- END ----------

CraigDC, 2023-07-19:
It doesn't crash when booting from USB, but it takes a while to boot.
I forgot to mention that the Nasher firmware you customized for me works fine with the Nasher ChromeOS.

MrChromebox, 2023-07-19:
what are the ChromeOS board names of these devices? I'm not seeing any problems here on WOLF or PANTHER

MrChromebox, 2023-07-19:
if the issue is only with Flex, then I'd say you should try using Brunch instead. Google specifically says Chromebooks are not supported with Flex

CraigDC, 2023-07-19:
I attempted to install the Google system-specific ChromeOS obtained through the Recovery Utility and it prompts me for a version. I tried all options, and nothing happens; just a black screen. I have installed the BIOS and OS on 60 Dell Chromebooks as per your original instructions. I assumed everything had installed OK, but at the end, I came to realize I had the crash and slow boot problem on the Dell's. The Acer's are fine and they have ChromeOS Flex. I'm pretty burnt out on this whole process.
Do you think Brunch will do the trick?
Thanks!

CraigDC, 2023-07-20:
"It" is  GNU GRUB. 
When I choose to boot from the USB stick to install the OS, it goes to GNU GRUB, which lists the following options to boot:
local image A
local image B
verified image A
verified image B
Alternate USB Boot

If I select Alternate USB Boot, I get the message "error: file "/boot/vmlinuz" not found. Press any key to continue"
any of the other options result in a black screen.
I attempted to install Windows and there were no problems.

MrChromebox, 2023-07-20:
You absolutely cannot boot/use a Google Recovery image with the UEFI firmware. Grub on a recovery USB is non-functional, it's leftover crap from ChromiumOS

CraigDC, 2023-07-23:
Well, I installed Brunch along with the Recovery Image as outlined here: https://www.reddit.com/r/Brunchbook/wiki/getting-started/#wiki_install_chromeos_from_linux_.28the_easiest_way.29
and it still crashes once and then boots; takes about 90 seconds to boot completely.

MrChromebox, 2023-07-23:
ok, well that's a brunch issue not a firmware one, sorry =(

CraigDC, 2023-07-26:
After using the Log Analyzer on the system log files since installing Brunch, it turns out the Chromebook isn't crashing, it's just booting twice.

CraigDC, 2023-07-26:
Does it matter that the Recovery Image version is 114 and the ChromeOS version is 115? There wasn't a 115 version of the Recovery Image.

MrChromebox, 2023-07-26:
I really don't know, I don't use Brunch and can't provide support for it. They have their own github and Discord server for that

closed 2026-11-04 by MrChromebox