MrChromebox/firmware #497: Asus CX5400 fails to detect nvme drives

luisjodiez, 2023-07-19:
Hi!

For some reason I have been unable to detect the nvme since I flashed the UEFI ROM.
My hardware is an i5/8GB/256GB model from [here](https://www.asus.com/laptops/for-home/chromebook/asus-chromebook-flip-cx5-cx5400/techspec/).

I disconected the battery to disable the RW protection.
I have flashed the UEFI ROM following [this](https://mrchromebox.tech/#fwscript).
I have tried to detect both stock and an 512GB m.2 nvme I had initialized with windows. Failed.

Any help or pointers would be apreciated.

MrChromebox, 2023-07-19:
what is unable to detect the NVMe? The firmware?

Can you boot from USB and get a cbmem log?

luisjodiez, 2023-07-20:
I think it's the firmware, yes. It's not mapped in the boot menu, and whenever I boot from a live usb doesn't show up.

Yeah, I can boot from a usb, but I don't know anything about cbmem logs. I'll update this comment as soon as I get something on this logs.

MrChromebox, 2023-07-20:
https://mrchromebox.tech/files/util/cbmem.tar.gz

download, extract, then run: `sudo ./cbmem -1 > cbmem.log` 
then attach the log file here (do NOT copy/paste it)

luisjodiez, 2023-07-20:
[cbmem.log](https://github.com/MrChromebox/firmware/files/12112237/cbmem.log)

There it is.

MrChromebox, 2023-07-20:
it appears that coreboot is not detecting the drive as present. Is this the stock NVMe drive? did it work with the stock firmware?

luisjodiez, 2023-07-21:
I think it was with the new drive, but just in case, I'm making a couple new cbmem logs, one for the new drive, one for the old one.
[cbmem-newdrive.log](https://github.com/MrChromebox/firmware/files/12117862/cbmem-newdrive.log)

MrChromebox, 2023-07-21:
neither drive shows up in the edk2 boot menu?

luisjodiez, 2023-07-24:
Sorry for the late reply.

No, the edk2 boot menu only shows the usb from which I live boot.

MrChromebox, 2023-07-24:
I'm going to need a cbmem from the stock firmware to compare then

luisjodiez, 2023-08-01:
I'll get it ready next week.

closed 2024-05-27 by luisjodiez