MrChromebox/firmware #499: No keypress event is being registered for the fullscreen key (BLOOGUARD chromebook)

apandada1, 2023-07-21:
I am using Arch Linux KDE with the MrChromebox.Tech full UEFI on a HP x360 14a (BLOOGUARD) chromebook.
I am using the Chromebook keyboard layout in KDE.

The fullscreen key on the keyboard is not working (everything else works). The same issue also present in Ubuntu MATE 22.04 and Kubuntu 23.04.

I tested that no keypress event is being registered using `xorg-xev`.

How to debug it?

The key was working in ChromeOS (so it is not a hardware issue).

MrChromebox, 2023-07-21:
the Google vivaldi keyboard driver translates the top row keyboard scancodes into media/action events - is that driver compiled as a module in your kernel? is it loaded? Do the other top row keys work?

apandada1, 2023-07-21:
Yes, the driver seems to be loaded. The driver is supposed to be present in Kernel 5.10 and beyond (I have 6.1.39 and 6.4.4)

```
archisman@archlinux:~$ sudo modinfo hid-vivaldi
filename:       /lib/modules/6.1.39-1-lts/kernel/drivers/hid/hid-vivaldi.ko.zst
license:        GPL
description:    HID vivaldi driver
author:         Sean O'Brien
srcversion:     4C225D6054BD74FFB33CCD2
alias:          hid:b*g0105v*p*
depends:        hid-vivaldi-common
retpoline:      Y
intree:         Y
name:           hid_vivaldi
vermagic:       6.1.39-1-lts SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        0E:C8:A6:5A:3B:71:E0:57:76:FA:14:02:D9:CB:C0:7B:7F:84:44:18
sig_hashalgo:   sha512
signature:      30:65:02:30:19:A9:8D:8B:70:AB:D6:B7:81:C6:C5:8A:17:87:BE:DA:
                F8:19:EF:D9:37:BA:A2:42:84:38:3C:14:72:D0:CC:85:7C:66:BE:F7:
                18:67:2A:6A:6B:88:25:21:58:D0:DE:7B:02:31:00:9E:F2:7E:2A:3F:
                C5:14:41:12:2A:8F:7F:04:40:D0:3D:53:57:8B:50:4E:BD:87:C2:F1:
                7F:44:09:4F:89:C3:17:8F:05:69:37:E0:F5:F8:EF:68:8B:45:BA:60:
                48:1E:26
archisman@archlinux:~$ sudo modinfo hid-vivaldi-common
filename:       /lib/modules/6.1.39-1-lts/kernel/drivers/hid/hid-vivaldi-common.ko.zst
license:        GPL
srcversion:     F44BC09AA9091A827A7214C
depends:        vivaldi-fmap
retpoline:      Y
intree:         Y
name:           hid_vivaldi_common
vermagic:       6.1.39-1-lts SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        0E:C8:A6:5A:3B:71:E0:57:76:FA:14:02:D9:CB:C0:7B:7F:84:44:18
sig_hashalgo:   sha512
signature:      30:65:02:31:00:C5:FF:63:56:AB:36:B0:56:37:F0:B6:CF:EC:E6:F8:
                03:AA:F3:81:1B:38:84:9D:3A:DD:39:AC:B2:D7:8D:ED:A0:19:81:8D:
                D2:89:34:2D:2F:5B:3F:8B:9F:99:55:83:54:02:30:59:E9:77:33:50:
                17:33:49:24:6C:99:38:D4:DD:6A:FE:C0:EF:00:B6:2F:5D:67:45:D8:
                FF:15:FF:74:9C:24:B5:43:0B:FD:F1:C0:2E:88:A0:71:8B:D4:B8:83:
                81:AF:65
```

All the top row keys (previous, next, reload, show-all-windows, brightness, sound, lock) work, but only the fullscreen does not work.

Also, in my other Chromebook (Acer c731 RELM) running Ubuntu MATE 22.04 (kernel 5.15) has all the keys working, but in the BLOOGUARD chromebook running Ubuntu MATE 22.04, the fullscreen key does not work. So, the kernel most likely has a working driver.

apandada1, 2023-07-21:
I also get
```
archisman@archlinux:~$ lsmod | grep vivaldi
vivaldi_fmap           16384  2 atkbd,cros_ec_keyb
```

MrChromebox, 2023-07-21:
you're not going to get a keypress event for any of the top row keys, since they use non-standard scancodes. Since all the other top row keys work, the driver is loaded, etc, it means that your OS/DE isn't handling the event generated. The firmware is doing the exact same thing as stock, only difference is the OS

apandada1, 2023-07-21:
In the affected chromebook, I am getting keypress events (in `xev`) for all other top row keys except fullscreen.

As I said, the same OS (Ubuntu MATE 22.04) gets keypress events for fullscreen (and others) for the other (RELM) chromebook, but it does not get keypress event (only for fullscreen) for this BLOOGUARD chromebook.

This seems more like a firmware rather than OS issue.

apandada1, 2023-07-21:
I don't know if this is relevant, but I found some keyboard related entries in journalctl.
```
Jul 21 08:30:23 archlinux kernel: cros-ec-keyb: probe of GOOG0007:00 failed with error -95
Jul 21 08:30:23 archlinux kernel: cros-ec-keyb GOOG0007:00: cannot register non-matrix inputs: -95
Jul 21 08:03:05 archlinux kernel: cros-ec-keyb: probe of GOOG0007:00 failed with error -22
Jul 21 08:03:05 archlinux kernel: cros-ec-keyb GOOG0007:00: cannot register non-matrix inputs: -22
```
It is always either 95 or 22.

MrChromebox, 2023-07-21:
RELM is a much older device, does not use vivaldi for top row mapping; RELM sends all the top row keys as F-keys. I'm guessing you changed the keyboard layout to Chromebook or the like?

apandada1, 2023-07-21:
Yes, I did change the keyboard layout to Chromebook

MrChromebox, 2023-07-21:
You don't need to / shouldn't do that on a device which uses Vivaldi. Set it back to default on BLOOGUARD and nothing will change I suspect

apandada1, 2023-07-21:
True. Fullscreen key does not, but everything else works the same way in the default keyboard layout in BLOOGUARD.

MrChromebox, 2023-07-21:
there's nothing different with the scancode being sent for fullscreen vs the other top row keys. It's a vivaldi driver issue or more likely OS event handler issue

apandada1, 2023-07-22:
I found that there is an advantage with Chromebook layout. `alt + backspace` works as `delete` in this layout, but that does not happen in the default layout.

> It's a vivaldi driver issue or more likely OS event handler issue

Can you suggest where I can file a bug report? Since Arch and Ubuntu are both affected, this is more of a kernel bug than a OS bug.

Anyway, thanks for your work. Even if this bug is not fixed, I can still happily live with the Chromebook.

Let's keep this issue open unless it is fixed in the kernel level.

MrChromebox, 2023-07-22:
I'm honestly not sure, looking at the vivaldi code (in the EC, in coreboot, and in the kernel), I don't see how there would be a problem with just the one key. the scancode for FULLSCREEN hasn't changed since it was implemented. I'm inclined to think it's a userspace issue

apandada1, 2023-07-23:
In case you have access to a recent Chromebook (does not have to be BLOOGUARD), can you check if there the fullscreen button works in Linux. I am curious if this is specific to BLOOGUARD, or happens in many recent Chromebooks using vivaldi.

MrChromebox, 2023-07-23:
I can reproduce the issue on another Chromebook using vivaldi for keyboard remapping

apandada1, 2023-07-27:
I found a workaround (thanks to WeirdTreeThing in  hultrabook in Discord).

First, install [keyd](https://github.com/rvaiya/keyd) (`yay -S keyd` in Arch).
Then, get the appropriate keyboard layout from [eupnea-utils](https://github.com/eupnea-linux/eupnea-utils/blob/main/configs/keyboard-layouts/cros-modern.conf). For most modern Chromebooks, [cros-modern.conf](https://github.com/eupnea-linux/eupnea-utils/blob/main/configs/keyboard-layouts/cros-modern.conf) should work.

Then, download the layout, and save it to `/etc/keyd/default.conf`.

```
cd /tmp
wget https://raw.githubusercontent.com/eupnea-linux/eupnea-utils/main/configs/keyboard-layouts/cros-modern.conf
sudo mv cros-modern.conf /etc/keyd/default.conf
```

Finally, enable and `keyd` with systemctl, and then, the fullscreen key should be working immediately.
```

apandada1, 2023-07-27:
@MrChromebox Can you add this to the your website so that users can find the workaround easily? I reproduced the same issue in another chromebook of my friend (`PHASER360S`), running Kubuntu 23.04.
Perhaps all modern Chromebooks that use Vivaldi are affected by this bug.

MrChromebox, 2023-07-27:
@apandada1 I don't see anything in cros-modern.conf that would change the function of the fullscreen key. 

I'd expect all vivaldi-enabled Chromebooks to behave the same way in this regard, they are sending the same scancodes

apandada1, 2023-07-27:
> I don't see anything in cros-modern.conf that would change the function of the fullscreen key.

Hmm. It maps `zoom` to `F11`. Why `zoom` (i.e. fullscreen button) did not get recognized before installing `keyd`, I have no idea about that.

Anyway, feel free to close this issue. However, it would be great if you mention the workaround in the main site because the issue is reproducible on possibly all vivaldi-enabled chromebooks, and this workaround seems to work.

Once again, thank you for your work.

closed 2023-11-05 by archisman-panigrahi