MrChromebox/firmware #503: HP Chromebook x360 14b (REV 4 MADOO) USB port not working after full bios flash

Qwer-TeX, 2023-08-10:
[HP Chromebook x360 14b-cb0013dx](https://support.hp.com/us-en/document/c07633606)

hello, I recently flashed a full uefi bios rom using the mrchromebox script and installed lubuntu on my 32GB eMMC drive 3 days ago and there were no issues.

But the very next day, I powered on the chromebook and found out the usb port doesn't work!
I was very very frustrated that the usb port doesn't work(because its the only usb port) so I though there might be some dust clogged under there that might be preventing me from using it, or probably the uefi bios wasn't providing the needed power voltage to the usb port that might've fried the usb port. Before I unscrewed my chromebook I plugged a usb hub to the usb port and the usb hub had a green blinking light.

So I unscrewed the back panel of my chromebook, unplugged the battery to find out that there was in fact no dust and the usb port wasn't even fried up. I screwed back my chromebook and the usb port started working. I plugged the usb hub and the green light became constant and didn't blink like how it should.

Next day, The same issue.
Unplug and replug the battery, usb port works!! why??? The battery isn't a removable battery, you have to unscrew the back panel to remove the battery. 

Not to mention before I flashed the full bios, I ran the bios by pressing Ctrl+L which booted linux from the usb port. I ran chromeos with Ctrl+D which booted ChromeOs, No usb port issues.

Can you fix this issue asap. I don't want to unplug and replug the non-removable battery every single day.

MrChromebox, 2023-08-10:
> Can you fix this issue asap. I don't want to unplug and replug the non-removable battery every single day.

then I'd recommend going back to stock for the time being. Probably an issue with the EC firmware update

MrChromebox, 2023-08-10:
shoot me an email or ping me on the chrultrabook Discord and I'll give you a test build with a different EC firmware

Qwer-TeX, 2023-08-11:
Yes I pinged you on discord

yankees5963, 2023-10-20:
I have the same laptop with a usb mouse plugged into that one usb port without issue.

closed 2026-11-04 by MrChromebox