MrChromebox/firmware #504: Acer BOBBA360 (CP311-2HN) brightness changes not observable

psstoyanov, 2023-08-11:
Hi,  I've installed the custom UEFI firmware on BOBBA360 device CP311-2HN with Fedora 38. I used EndevourOS img for secondary check.

Everything seems to work supririsngly well for daily operations with the exception of the brightness changes not being applied.
It appears as if the values are changing but there is no observable difference on the unit's dispay.
The same behaviour was observed on Gnome with Fedora 38 and XFCE with EndevourOS.

Has someone else observed similar behaviour?

MrChromebox, 2023-08-12:
I've not had reports of any GLK devices having issues with brightness control. your best bet for assistance is probably the chrultrabook Discord linux channel. They can help verify the correct backlight control type is being used

psstoyanov, 2023-08-14:
So far no movement in the Discord channel.
The values are indeed changing in `/sys/class/backlight/intel_backlight/brightness` as expected but nothing observable on the unit.

I'm tempted to check with Tiny11 to verify if its Linux only or not.

MrChromebox, 2023-08-14:
Under Windows, does the backlight adjustment slider exist and not work, or not exist at all

psstoyanov, 2023-08-14:
It exists but doesn't work.

I didn't bother to check why the keyboard brightness buttons, touchpad and audio don't work. To adjust the brightness I've went through Settings -> Display

MrChromebox, 2023-08-14:
that most likely means the backlight control type for the display is incorrect in the VBT in the firmware. Does your device have an OLED panel ?

MrChromebox, 2023-08-14:
boot linux with the following kernel param added: drm.debug=0xe

then get me the output of `sudo dmesg | grep drm` so I can see what's going on with the backlight detection/control

psstoyanov, 2023-08-15:
Applied `drm.debug=0xe` to `/etc/default/grub` :

<details>

<summary>/etc/defailt/grub </summary>

```
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR="$(sed 's, release .*$,,g' /etc/system-release)"
GRUB_DEFAULT=saved
GRUB_DISABLE_SUBMENU=true
GRUB_TERMINAL_OUTPUT="console"
GRUB_CMDLINE_LINUX="rhgb drm.debug=0xe"
GRUB_DISABLE_RECOVERY="true"
GRUB_ENABLE_BLSCFG=true
```
</details>

Result from `dmesg | grep drm`:
<details>
<summary> dmesg | grep drm </summary>

```ShellSession

[    4.069119] [drm:drm_client_modeset_probe] connector 144 enabled? yes
[    4.069121] [drm:drm_client_modeset_probe] connector 152 enabled? no
[    4.069123] [drm:drm_client_modeset_probe] connector 163 enabled? no
[    4.069128] [drm:drm_client_modeset_probe] Not using firmware configuration
[    4.069130] [drm:drm_client_modeset_probe] looking for cmdline mode on connector 144
[    4.069133] [drm:drm_client_modeset_probe] looking for preferred mode on connector 144 0
[    4.069135] [drm:drm_client_modeset_probe] found mode 1366x768
[    4.069137] [drm:drm_client_modeset_probe] picking CRTCs for 1366x768 config
[    4.069140] [drm:drm_client_modeset_probe] desired mode 1366x768 set on crtc 68 (0,0)
[    4.083601] i915 0000:00:02.0: [drm:drm_mode_getconnector] User-space requested a forced probe on [CONNECTOR:144:eDP-1] but is not the DRM master, demoting to read-only probe
[    4.083672] i915 0000:00:02.0: [drm:drm_mode_getconnector] User-space requested a forced probe on [CONNECTOR:152:DP-1] but is not the DRM master, demoting to read-only probe
[    4.083685] i915 0000:00:02.0: [drm:drm_mode_getconnector] User-space requested a forced probe on [CONNECTOR:163:DP-2] but is not the DRM master, demoting to read-only probe
[    4.091837] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:169]
[    4.120623] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:169]
[    4.131751] [drm:drm_mode_setcrtc] [CRTC:68:pipe A]
[    4.131777] [drm:drm_mode_setcrtc] [CONNECTOR:144:eDP-1]
[    4.153153] [drm:drm_helper_probe_single_connector_modes] [CONNECTOR:144:eDP-1]
[    4.153175] i915 0000:00:02.0: [drm:intel_dp_detect [i915]] [CONNECTOR:144:eDP-1]
[    4.153429] i915 0000:00:02.0: [drm:intel_dp_detect [i915]] [ENCODER:143:DDI A/PHY A] MST support: port: no, sink: no, modparam: yes
[    4.153641] i915 0000:00:02.0: [drm:intel_dp_print_rates [i915]] source rates: 162000, 216000, 243000, 270000, 324000, 432000, 540000
[    4.153851] i915 0000:00:02.0: [drm:intel_dp_print_rates [i915]] sink rates: 162000, 270000
[    4.154102] i915 0000:00:02.0: [drm:intel_dp_print_rates [i915]] common rates: 162000, 270000
[    4.155639] i915 0000:00:02.0: [drm:update_display_info.part.0] [CONNECTOR:144:eDP-1] Assigning EDID-1.4 digital sink color depth as 6 bpc.
[    4.155652] i915 0000:00:02.0: [drm:update_display_info.part.0] [CONNECTOR:144:eDP-1] ELD monitor 
[    4.155657] i915 0000:00:02.0: [drm:update_display_info.part.0] [CONNECTOR:144:eDP-1] ELD size 20, SAD count 0
[    4.155667] i915 0000:00:02.0: [drm:intel_dp_set_edid [i915]] [CONNECTOR:144:eDP-1] VRR capable: no
[    4.155944] i915 0000:00:02.0: [drm:intel_dp_set_edid [i915]] [CONNECTOR:144:eDP-1] DFP max bpc 0, max dotclock 0, TMDS clock 0-0, PCON Max FRL BW 0Gbps
[    4.157891] i915 0000:00:02.0: [drm:intel_dp_set_edid [i915]] PCON ENCODER DSC DPCD: 00 00 00 00 00 00 00 00 00 00 00 00 00
[    4.158174] i915 0000:00:02.0: [drm:intel_dp_set_edid [i915]] [CONNECTOR:144:eDP-1] RGB->YcbCr conversion? no, YCbCr 4:2:0 allowed? no, YCbCr 4:4:4->4:2:0 conversion? no
[    4.159110] [drm:drm_helper_probe_single_connector_modes] [CONNECTOR:144:eDP-1] probed modes :
[    4.159117] [drm:drm_mode_debug_printmodeline] Modeline "1366x768": 60 76300 1366 1414 1446 1600 768 772 778 794 0x48 0xa
[    4.159189] [drm:drm_helper_probe_single_connector_modes] [CONNECTOR:152:DP-1]
[    4.159195] i915 0000:00:02.0: [drm:intel_dp_detect [i915]] [CONNECTOR:152:DP-1]
[    4.159419] [drm:drm_helper_probe_single_connector_modes] [CONNECTOR:152:DP-1] disconnected
[    4.159431] [drm:drm_helper_probe_single_connector_modes] [CONNECTOR:163:DP-2]
[    4.159436] i915 0000:00:02.0: [drm:intel_dp_detect [i915]] [CONNECTOR:163:DP-2]
[    4.159654] [drm:drm_helper_probe_single_connector_modes] [CONNECTOR:163:DP-2] disconnected
[    6.082715] systemd[1]: Starting modprobe@drm.service - Load Kernel Module drm...
[    6.190617] audit: type=1130 audit(1692084768.619:10): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=modprobe@drm comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    6.190731] audit: type=1131 audit(1692084768.619:11): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=modprobe@drm comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    7.214906] i915 0000:00:02.0: [drm:intel_pps_vdd_off_sync_unlocked [i915]] [ENCODER:143:DDI A/PHY A] PPS 0 turning VDD off
[    7.215175] i915 0000:00:02.0: [drm:intel_pps_vdd_off_sync_unlocked [i915]] [ENCODER:143:DDI A/PHY A] PPS 0 PP_STATUS: 0x80000008 PP_CONTROL: 0x00000067
[    9.128842] i915 0000:00:02.0: [drm:intel_backlight_device_update_status [i915]] updating intel_backlight, brightness=19200/19200
[    9.129137] i915 0000:00:02.0: [drm:intel_panel_actually_set_backlight [i915]] [CONNECTOR:144:eDP-1] set backlight level = 19200
[   10.806867] i915 0000:00:02.0: [drm:intel_power_well_enable [i915]] enabling PW_2
[... cut ...]
[   74.390627] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:171]
[   74.722537] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:174]
[   74.755998] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:171]
[   74.789648] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:174]
[   74.822637] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:171]
[   74.856588] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:174]
[   74.889493] i915 0000:00:02.0: [drm:drm_mode_addfb2] [FB:171]
[   74.922456] i915 0000:00:02.0: [drm:drm_mod

psstoyanov, 2023-08-15:
Where are these firmware blobs fetched from?
- https://github.com/MrChromebox/coreboot/blob/ddaa6478511fb88e51957681dd1462eac0f3c3d7/configs/glk/config.bobba360.uefi#L14

MrChromebox, 2023-08-15:
@psstoyanov fetched from? They are built by me, and stored in the blobs repo as per the path

MrChromebox, 2023-08-15:
no, they don't exist in any repo. Google builds them on the fly and includes them in the FW_MAIN_A/B regions. I build from the same branch but latest version and with a few tweaks

psstoyanov, 2023-08-19:
Interesting, `ectool` throws invalid command:
```
$ sudo ./ectool backlight 0
EC result 1 (INVALID_COMMAND)
```

MrChromebox, 2023-08-19:
that's what I'd expect for a device without a keyboard backlight. The EC has nothing to do with the screen backlight.

psstoyanov, 2023-08-19:
This is confusing. Why is there LCD reference in the command? :confused: 
https://chromium.googlesource.com/chromiumos/platform/ec/+/master/util/ectool.c#74

MrChromebox, 2023-08-19:
maybe some ARM-based devices use the EC for backlight control? Intel/AMD ones do not

psstoyanov, 2024-06-16:
Alright some further progress. After I haven't used the device for.... nearly a year.
What I find interesting is that the digitizer also gave the ghost rendering the touch functionality unusable. The Wacom stylus is still functional.

As a reminder, the original issue highlighted that I couldn't control the brightness of the LCD apart from completely shutting it down.

### First steps
I've returned the device to stock firmware with the stock image (which should override with default HWID) and the backlight adjustment doesn't work there as well. My initial feeling was something about the HWID interfering with the correct hardware behavior.

### Second try with stock firmware
Moving back and forth (as the tool can't override the original firmware while the system is using CrOS) - the original firmware is now back on the device.

### Results
Results of using the original firmware with correct HWID:
- Backlight control is still not functional.

### Further findings
Looks like something gave the ghost. What exactly I'm not sure of. 
After some additional observation of the hardware, I can see "bubbles" forming in the lower part of the display (right where the data/ backlight connection of the LCD is behind the lid). 
<details>
<summary> The bubbles and their location behind the LCD </summary>

[image]

[image]

</details>

I suspect the relative who owned the device, induced water damage that slowly crept it's way through the unit (not sure if this was even registered as it was at least partially functional or functioning properly for some time).

I will send the device somewhere for further investigation or order a new display assembly.

### Conclusion
In conclusion, using the custom firmware acts **exactly** as expected and **none** of the problems with this particular BOBBA360 have any connection to the scripts and the custom firmware.

@MrChromebox thank you for the support and for the incredible tool you've created! Once this little bobba360 is mended, you can be sure I will use the tool again 💯

closed 2024-06-16 by psstoyanov