MrChromebox/firmware #506: Google_Endeavour Support?

mdanielbelarmino, 2023-08-15:
I want to turn my ChromeOS device into WindowsOS. Currently, it is being used for Google Meet Room. I was able to ran the firmware utility script, but the device and platform is unrecognized. Because of that, UEFI install option was disabled. Please see attached image for reference. The device is Google Meet Series One which has i7 8th gen processor(I also have the i7 13th gen processor). Will there be a support for this machine in the future? Thank you so much Mr. ChromeBox. :)
[image]

MrChromebox, 2023-08-15:
I'll try to add support in the next release. If you have a Suzy-Q cable and/or are ok flashing a test build, shoot me a msg on Discrd

mdanielbelarmino, 2023-08-15:
Thanks for the quick reply. I will purchase a Suzy-Q cable since it is needed, I will let you know if I got one. Thanks again.

mdanielbelarmino, 2023-08-16:
I sent the files to your email. Let me know you have further favor, I am happy to help. thanks

MrChromebox, 2023-08-18:
endeavour has been tested working and will be added in the upcoming 4.21 release

mdanielbelarmino, 2023-08-21:
Thanks MrChromebox. I will now close this ticket.

closed 2023-08-21 by mdanielbelarmino

BringusStudios, 2023-11-01:
Sorry to reopen this, but I have the same Google Meet endeavour device, and I am not able to remove the fw write protect on it. How were you able to do it?

MrChromebox, 2023-11-01:
@BringusStudios a SuzyQ cable is needed, best I can tell

BringusStudios, 2023-11-01:
> @BringusStudios a SuzyQ cable is needed, best I can tell

I've got a homemade SuzyQ that should be working, but it won't show up under dmesg and there's no /dev/ttyUSB device either when I plug it in in either orientation. Also, in the first image that he had sent, he has fw wp disabled already, before you had told him to get a SuzyQ cable. I'm not sure if it's even necessary for this computer. I'm working on getting a Chromebook with type C to validate my cable right now.

MrChromebox, 2023-11-01:
it's a Kabylake board, it definitely supports CCD/CR50 unlock

burritoheada, 2023-12-12:
so i had issues with ur firmware script as it wont let me download the full room and it says my device is undetectable and smth

MrChromebox, 2023-12-12:
@burritoheada you're using an ENDEAVOUR board? If not, then start a new issue and provide the necessary detail for me to help.

burritoheada, 2023-12-12:
I created a new issue

vivithecanine, 2024-01-08:
Closing the loop on this on GitHub so things don't get lost.

On Genesis, and very likely Endeavor, if the screw terminal contacts are bridged on the board, hardware WP is disabled on the SPI flash.

Using a Suzyquable does not seem to expose UART in any orientation on these platforms, even after setting CCD Privilege level to `Open`.