MrChromebox/firmware #507: FEATURE REQUEST: Menu option to remap left alt key to Fn key

atari2600a, 2023-08-16:
I installed arch & KDE Plasma on a Samsung 4 chromebook.  You know, the macbook clone with 4GB of RAM.  The function keys work out of box but not the 'function keys'.  I recall the seabios builds included this back when I was messing around with the arguably better C710's.  But I wanted a sturdier build this time around.  &, it's like, it's the one thing keeping me from having the *perfect* thin client solution.  Seriously, everything is working & I have an IPS panel on the way help me have this one win I'm too laaaazy to figure out how the menu subsystem works, how the HID implementation works, etc.  If no one takes this up, I'll make it my weird....thing.  Make some commits to a clone with modifications extending all the way down to the buildflags.

MrChromebox, 2023-08-16:
there is no function key on Chromebook keyboards, so there is no function modifier in the EC keyboard code. You'd have to make a non-trivial amount of changes to the EC to make this happen.

atari2600a, 2023-08-22:
Okay so, I've *tried* to actually dig into this a bit.  You know the deal, the industry likes to obfuscate the documentation for things that control things...that control things *just* enough to prove your wholehearted dedication.
     Are we talking the RO burnin portion?  If I were actually crazy enough to learn, say, zed 80 (gameboy) programming just to have an architectual baseline, then work up to embedded x86 programming in C on a higher level of learning, then finally try to dive in again, is there anything preventing this from being an interrupt in the rw portion?

Then I tried this uinput based daemon called ydotool but I couldn't get it to communicate with itself despite digging through all the docs for all the pieces.  I don't think it knows how to tap the cros-ec-keyb module.  That or my service definitions are terrible.

     I know things have changed a lot but I'm still working on the original structure from that one "we invented a new platform & we're launching a new product sorry for google wave" keynote.  You know the one, the one where the journalists were having a stroke asking questions & the speakers kept getting very noticably pissed that they didn't understand the high-to-low level concepts they just spent 2 hours explaining.

atari2600a, 2023-08-22:
BTW, any way to get secure boot into setup mode?  Also, there's this weird glitch where it won't show certain confirmation menus without first going to & exiting efishell.  I should have probably made that an actual bug report...

MrChromebox, 2023-08-23:
you'll need to change the EC-RW firmware, not the EC-RO. The EC code is just standard C, nothing special. And the keyboard handler isn't super complex.

MrChromebox, 2023-08-23:
> BTW, any way to get secure boot into setup mode? Also, there's this weird glitch where it won't show certain confirmation menus without first going to & exiting efishell. I should have probably made that an actual bug report...

I'm not sure, I'll have to look into this

cameraphone77, 2024-04-23:
I had the same problem with my flex 5i lillipup (no function keys), simple solution is install keyd (start and enable service) and add the following file to /etc/keyd/default.conf

```
[ids]

*

[meta]

back=f1
forward=f2
refresh=f3
zoom=f4
scale=f5
brightnessdown=f6
brightnessup=f7
mute=f8
volumedown=f9
volumeup=f10
sleep=f11
```

Now when you hold down the meta (search) key and press back it's the same as pressing f1.
switching VT is:  Search+ctrl+alt+back

closed 2026-11-04 by MrChromebox