MrChromebox/firmware #509: Dell Inspiron 14 7486

osterac, 2023-08-19:
Hi, I am working on a Dell Inspiron 14 7486 and I noticed there is no firmware available for it, although there is firmware for the 7460. What would be required to add support for this model? The owner wants to put a different operating system on it for games and etc. If required I could ask if he is willing to loan it to you.
Thanks
Allen

MrChromebox, 2023-08-19:
as painfully detailed on the supported devices page, support is not determined by make/model, but but ChromeOS board name. I'd be willing to bet it's the same for the 7480 and 7486, but you'll need to verify that yourself.

Please re-open if the board name differs from one listed on on the supported devices page

closed 2023-08-19 by MrChromebox

osterac, 2023-08-19:
Sorry about that. The only reason I posted was because I got an error from the script saying my chromebook wasn't compatible. I ran the script just now and it worked - I don't know if that is because I downgraded to chomeos 110 or if I just did it wrong before.
the 7486 is Vayne by the way.

MrChromebox, 2023-08-19:
not sure why you'd have gotten an unsupported error, VAYNE has been supported since 2017

osterac, 2023-08-22:
The boys did work well for me, I tried about every Linux distro under the sun but there just isn't any audio support for my Chromebook yet so I ended up using crouton and auto starting a chroot of Debian bullseye on startup. ALMOST as good.

ShapeShifter499, 2023-08-22:
@osterac I happened to see this pop up in my email.  Have you checked out https://github.com/WeirdTreeThing/chromebook-linux-audio ?

It may help, or you can open an issue ticket there. The dev for that project has had great success getting audio working on many different chromebooks.

osterac, 2023-08-22:
I did try that, didn't work, according to the compatibility chart my model is a work in progress. Thanks though.