MrChromebox/firmware #511: Booting Issue on HP Chromebook 11 (Kip Board)

BuddyBadger, 2023-08-22:
Hello all, I have two HP Chromebook 11s (which have the Kip Board) that I am trying to convert over to Linux. I have removed the write protect screw and flashed the firmware following the instructions on MrChromebox's website. 

Everything went well, and I was able to test install two different Linux Distros. However now, two days later, I cannot get the Chromebook to boot any more. 

When I hit the power button, the screen comes on blank. Then after about a minute, the screen's back light start to flash on and off in intervals. The only way to get it to stop is to hold down the power button.

I immediately assumed that something must have gone wrong in the flashing process and got ready to follow the bricked device instructions. However after turning it on and off again multiple times, it randomly actually booted to UEFI. In Lubuntu, I reran the MrChromebox's script and reinstalled the firmware. 

I turned the device off and back on and it booted fine into Lubuntu. I turned it off to install a new OS and now it is booting to a black screen before eventually flashing again.

Any help is greatly appreciated and let me know if I can provide any other information that may be useful.

MrChromebox, 2023-08-22:
this is very odd, not sure what to suggest here unfortunately

BuddyBadger, 2023-08-22:
> this is very odd, not sure what to suggest here unfortunately

Well shoot... Do you think getting a CH341a programmer and manually re-flashing might help or do you think it may be a lost cause atm?

MrChromebox, 2023-08-23:
I don't think it would hurt, but I'm not optimistic it would help

donsimiondon, 2025-08-07:
The same thing is happening to me on a Lenovo Chromebook. Were you able to resolve the issue?