MrChromebox/firmware #512: TEEMO - Asus Chromebox 3 - replace wifi card by second ethernet adapter

namezys, 2023-08-23:
I want to add 2.5Gb ethernet card to my chromebox. I added it and it doesn't work.

firmware-util.sh output:
```
**   Device: Asus Chromebox 3 / CN65 (TEEMO)
** Platform: Intel KabyLake
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.20.1 (07/21/2023)
**    Fw WP: Disabled
```

namezys, 2023-08-23:
There are outputs for:
* cbmem
* lspci (with -v and without)
* dmesg
[asus_chromebox_3.zip](https://github.com/MrChromebox/firmware/files/12423958/asus_chromebox_3.zip)

namezys, 2023-08-24:
Some additional info

From here: `variants/baseboard/devicetree.cb`

```
		# PCI Express Port 3 for LAN, will be swapped to port 1 by FSP
		device pci 1c.2 on
			chip drivers/net
				register "customized_leds" = "0x0fa5"
				register "wake" = "GPE0_PCI_EXP"
				device pci 00.0 on end
				register "device_index" = "0"
			end
		end # PCI Express Port 3
		device pci 1c.3 on
			chip drivers/wifi/generic
				register "wake" = "GPE0_PCI_EXP"
				device pci 00.0 on end
			end
		end # PCI Express Port 4 for WLAN
```

I looked into `src/device/pci_device.c`

As I undertand, `pci_probe_dev` expect that there is a WiFi device on `1c.3`.

I suppose that I can remove chip section:
```
		# PCI Express Port 3 for LAN, will be swapped to port 1 by FSP
		device pci 1c.2 on
			chip drivers/net
				register "customized_leds" = "0x0fa5"
				register "wake" = "GPE0_PCI_EXP"
				device pci 00.0 on end
				register "device_index" = "0"
			end
		end # PCI Express Port 3
		device pci 1c.3 on end # PCI Express Port 4 for WLAN
		device pci 1c.4 on end # PCI Express Port 5 for NVMe
```

In this case I can lost ability to `wake on lan` but it acceptable for me.

To be honest, I don't know how to build everything. If you have a build script, it would be very helpful.

MrChromebox, 2023-08-24:
> As I undertand, pci_probe_dev expect that there is a WiFi device on 1c.3.

that is not correct. The PCIe port is enabled, and that's it. the comment is simply to indicate what device is attached / why it's enabled. pci_probe_dev has no expectation as to what device is going to be connected to the root port.

coreboot is simply failing to detect any device attached to the port, for reasons unknown. There are no changes which we can make to the devicetree/overrridetree which will change that behavior. I'd likely need to add some more debug output to coreboot to determine the source of the issue

tirsojrp, 2023-08-24:
Already tried... my card is a 2.5G Intel i225 v3 (M key). Tried on the NVME/SATA port and the WiFi port with an adapter. It works on any other pc I have tried.

MrChromebox, 2023-08-24:
already tried...what?

your other PC may have the the m.2 sockets wired differently so it's not directly comparable to that on TEEMO

is this similar to what you're trying to use?
www.amazon.com/dp/B0BPCT87MS

namezys, 2023-08-24:
This is the card that I have:
https://vi.aliexpress.com/item/1005005101361503.html

namezys, 2023-08-24:
I want to add more logs to open core.
@MrChromebox just to confirm that I can build ROM using your script:
[coreboot](https://github.com/MrChromebox/coreboot/tree/MrChromebox-4.20)
/build-uefi.sh

Maybe I'm not correct (I started to dig into opencore code yesteday) but I suggest that:
* `pci_probe_dev` gets preconfigured `dev`
 * because there is a static config
 * and because this log message only here
* therefore it goes to `else` branch and it can not read `PCI_VENDOR_ID`
* so it return `null`

BTW, if I can build it, I would be able to debug it.

```
struct device *pci_probe_dev(struct device *dev, struct bus *bus,
				unsigned int devfn)
{
	u32 id, class;
	u8 hdr_type;

	/* Detect if a device is present. */
	if (!dev) {
		struct device dummy;
                ...
		id = pci_read_config32(&dummy, PCI_VENDOR_ID);
                ...
		dev = alloc_dev(bus, &dummy.path);
	} else {
                ...
		id = pci_read_config32(dev, PCI_VENDOR_ID);

		/*
		 * If the device does not have a PCI ID disable it. Possibly
		 * this is because we have already disabled the device. But
		 * this also handles optional devices that may not always
		 * show up.
		 */
		/* If the chain is fully enumerated quit */
		if ((id == 0xffffffff) || (id == 0x00000000) ||
		    (id == 0x0000ffff) || (id == 0xffff0000)) {
			if (dev->enabled) {
				printk(BIOS_INFO,
				       "PCI: Static device %s not found, disabling it.\n",
				       dev_path(dev));
				dev->enabled = 0;
			}
			return dev;
		}
	}
       ...

	/* Display the device. */
	printk(BIOS_DEBUG, "%s [%04x/%04x] %s%s\n", dev_path(dev),
	       dev->vendor, dev->device, dev->enabled ? "enabled" : "disabled",
	       dev->ops ? "" : " No operations");

	return dev;
}
```

tirsojrp, 2023-08-24:
> already tried...what?

Tried to add a second ethernet adapter to a CN65.

> your other PC may have the the m.2 sockets wired differently so it's not directly comparable to that on TEEMO

Of course, tested to make sure card and adapter were working properly.

> is this similar to what you're trying to use?
> www.amazon.com/dp/B0BPCT87MS

Different ethernet port breakout, but looks the same.

https://www.ebay.com/itm/204175573191

MrChromebox, 2023-08-24:
Ubuntu 2204 works fine. you need to properly set up the coreboot toolchain etc and clone submodules as per the setup instructions on coreboot.org. CPU microcode is included in the blobs submodule and will automatically be included in the build once the submodule is cloned

namezys, 2023-08-24:
Thank you. It works on 22.04. Also, I looked into the build instructions for edk2 to add other packages.

namezys, 2023-09-01:
Hey.

I have a great news:
```
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
03:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983
```

I built dozens of firmware with a lot of debugging and understood a lot about PCI probes/scans. However, nothing related to an attached device.

Next step I looked at the WiFi card and checked the pinout of M-2 A+E port.

[image]

One pin looked very promising: CLKREQ.

Next was RTL8125.

[image]

Here this pin is unconnected therefore it's tri-state.

From some documentation (short description of PCI-E):
```
The card must drive the CLKREQ# signal low during power up, whenever it is reset, and whenever it requires the reference clock to be in the active clock state.
```

As a next step, I tapped this pin on the WiFi card. After boot this card is dissapered from PCI devices.

And final step - just connect this to ground pin (right to CLKREQ#) on ethernet card.

[image]

And it is in device list right now.

Next task: 
* try to find if it's possible to disable checking CLKRTL# in firmware.

And I can confirm that there is only PCI-E x1

MrChromebox, 2023-09-02:
> try to find if it's possible to disable checking CLKRTL# in firmware.

you can enable/disable/adjust CLKREQ for PCIe devices, see:
/home/mattd/dev/coreboot/src/mainboard/google/fizz/variants/baseboard/devicetree.cb starting around line 161

namezys, 2023-09-02:
Do you mean `register "PcieRpClkReqSupport[3]" = "1"`.

I tried it. Looks like it's affects only devices and does not affect bus initialisation.
So you can disable this feature on add-on card only.
I should find the place there PCI bus initialisation.
Also I'm looking where PCI lanes combined to links. I hope I can turn NVMe slot (x4) to two (2x2) PCI slots.

MrChromebox, 2023-09-02:
> Also I'm looking where PCI lanes combined to links. I hope I can turn NVMe slot (x4) to two (2x2) PCI slots.

that's controlled by the soft straps in the IFD, you need Intel's FIT tool to configure, and flash externally

namezys, 2023-09-03:
@MrChromebox 
Thank you. If you have something to start, it would be great. I found few information but helpful would be appreciate.
However, this investigation will be much slower.

Also, I check mini-pci SATA that I tried to use on my previous Chromebox. Looks like same issue. unfortunately, I don't have this old Chromebox any more and can't check.

[image]

[image]

I suppose that it would be great to write short manual how to enable this add-on card.

MrChromebox, 2023-09-03:
look at the Win-raid forums for the download you'll need for the FIT tool

namezys, 2023-10-08:
Hey, guys.

Win-raid is very helpful.

Right now I found how to disable checking CLKREQ.

[image]

Continue to play around.

windsifter, 2023-11-18:
Did you ever get it working? I was tempted to try and add a 2.5Gb ethernet adapter there as well, but if the hardware requires special software, I'd be at a loss. Did you punch another hole out the back or side for the ethernet cable?

namezys, 2023-11-18:
> Did you ever get it working? 
Yes. It's actually easy.

> I was tempted to try and add a 2.5Gb ethernet adapter there as well
We did it (Realtek 2.5Gb) and it works better than embodied RTL controller

> if the hardware requires special software
Nothing special, this is general PCI bus so you need only driver for your card (I suppose that you have Realtek or Intel)

> Did you punch another hole out the back or side for the ethernet cable?
Right now it's open but we want to print a new case. So it is not a big issue for us. 

--------

PS: I'm thinking about bifurcation of NVMe but I need an adapter. It can be ordered from GLS PCB but the cost of one will be about $2-3 in case of order about 10 pieces and $20 in case of only one

wicadmin, 2024-07-21:
@MrChromebox  - is there a built-in firmware way of doing this now or do I still need to bridge that pin to ground? I'm on Kuldax and facing the same issue with an intel i226-v A+E m.2 2.5Gbps NIC card.

namezys, 2024-07-21:
@wicadmin there is no way to modify this area of firmware from software, the only one way is to flash directly with programmer. And no general way to modify it - each case is unique and can depend on small hardware changes like revision. 

Just as bridge between two pins with conduction ink and it will be enough.

wicadmin, 2024-07-21:
@namezys - thanks. Do you know if just masking off the CLK REQ pin with plastic tape would work? I had done this with an HBA storage card a while back.

namezys, 2024-07-21:
@wicadmin the slot is too tight… so usually impossible

namezys, 2024-07-21:
Oops. You can’t masked off this pin, it must be pull down to ground.

wicadmin, 2024-07-21:
> Oops. You can’t masked off this pin, it must be pull down to ground.
Yep, masked it and still it didn't show up :-)

OK, trying with what conductive paint I have. I found some cold galvanized paint (zinc only). See if it conducts when dried.

wicadmin, 2024-07-21:
well, apparently the pin next to what is supposed to be the CLK REQ is not a ground pin (Only the tall square top pins are ground)

[image]

wicadmin, 2024-07-21:
Well, I overlayed the card that came with the chromebox (e-key) but most of the PCI pins not lining up. I assume this may be part of the issue.

[image]

namezys, 2024-07-22:
@wicadmin I'm sure that this misplacement should not be an issue (95%, of course stupidness of engineer is possible).

However, It looks like SATA card. Missing pins are 2 data stream (up and down) and clock.
I'm always mixed up different slots but I feel than 4 couples of missing pins are:
* data up: 2 pins
* data down: 2 pings
* clock: 2 pins
* click req
* and something like WA enable

Also pairs of data and clock is separated by one ground. 

Can you send pics of all cards, maybe I will be more precise because we are working on NSA for home and I read a lot.

wicadmin, 2024-07-22:
@namezys - these are the photos. The orginal wifi/bluetoothj combo that came with the chromebox and the A+E 2.5Gbps nic.

[image]

[image]

wicadmin, 2024-08-02:
@namezys  So, I returned the i226 card and bought a RTL. However, although it looks completely like your's, the CLRCLK pin is connected via PCB trace. I however do not see it appear when the app chromebox is booted up.

[image]

MrChromebox, 2024-08-02:
if it's a #CLK_REQ issue, can either disable use or set to free-run in the firmware, then rebuild

wicadmin, 2024-08-02:
> if it's a #CLK_REQ issue, can either disable use or set to free-run in the firmware, then rebuild

Thanks. I haven't build by own before. I'll see if @namezys  has any ideas and if not I'll attempt to do a rebuild.

Zema4You, 2024-08-17:
Thank You @namezys, Your solution with soldering CLKREQ to GND works for me too.

wicadmin, 2024-08-17:
> Thank You @namezys, Your solution with soldering CLKREQ to GND works for me too in other mother boards.

@Zema4You - for the m.2 cards with CLKREQ connected like the one I recently posted in this thread, you were able to get working? Was there anything you needed to do in EFI etc? That one above (the Realtek, I returned the Intel since the pinout seemed different) is also not being detected.

Zema4You, 2024-08-17:
Soldered 1GbE RTL card:
[image]

Zema4You, 2024-08-17:
Working Intel I210 card without soldering:
[image]

Zema4You, 2024-08-17:
This cards work for me - they are visible by PCI command in UEFI Shell and works in Windows and Linux.

wicadmin, 2024-08-17:
> This cards work for me - they are visible by PCI command in UEFI Shell and works in Windows and Linux.

Thanks. Something strange is happening with the one that has the pin connected on the trace to the chip. Not sure if to break that trace and solder to ground.

maciel310, 2024-08-30:
Has anyone had any luck with the ASM1166-based "M" key 6x SATA port controller (specifically this one: https://www.aliexpress.us/item/3256805467391305.html)? I tried replacing the NVME drive with it (not the WiFi board), but no drives are detected in Linux and it doesn't show up in `lspci`. I tried it with both Debian and SystemRescueCD. I confirmed that the controller and HDD work by putting it in my desktop and it works as expected there.

I haven't tried messing with the CLKREQ pin since this is replacing the NVME and not the WiFi, and if I'm looking at things correctly it doesn't look like CLKREQ is floating on this PCB (there's a trace leading away from that pin). I did have to find a pinout for the M key since it looks like that's different than the A+E key examples posted above.

Any tips on debugging this further would be appreciated!

maciel310, 2024-09-06:
Nevermind, I just re-tested this and it looks like it is working just fine. Not sure exactly what changed, perhaps it was a loose connection or something along those lines (though I did re-seat it during my initial testing).

If anyone wants to get one of these, just keep in mind that the heatsink that comes with it will barely interfere with the second m.2 slot. It's easy enough to remove if you need both, and you can reattach it moved over a few mm and get the clearance you need.

MrChromebox, 2026-11-05:
I'm adding a UEFI setup menu option to disable CLKREQ for the WiFi socket in case the adapter card doesn't support it, anyone want to test? this means no more pin grounding necessary.

ref: https://review.coreboot.org/c/coreboot/+/97068