MrChromebox/firmware #516: chromebook barla full UEFI firmware: cold boot archlinux ok, soft boot doesn't recognize internal storage

ccordero90, 2023-09-10:
hello

I flashed full UEFI firmware on my barla chromebook uneventfully then proceded to install archlinux to internal storage which also worked up ok
but when i restart the computer (soft boot) UEFI shell does not recognize internal storage, not being able to boot the OS, not even with the boot from file option, it does recognize external drives and mmc cards though

this issue does not appear when shutting down computer (cold boot), booting to OS occurs normally

ty in advance

MrChromebox, 2023-09-10:
known issue on Stoneyridge devices

GamePlayer-8, 2023-10-14:
At the moment the alternative solution is kexec reboot.

Example:
```shell
#!/bin/sh

reuse="--reuse-cmdline"
kernel="${kernel:-$(uname -r)}"

if [ -f "/boot/vmlinuz-linux" ]; then
	kargs="/boot/vmlinuz-linux --initrd=/boot/initramfs-linux.img"
else
	kargs="/boot/vmlinuz-$(kernel) --initrd=/boot/initramfs-$(kernel).img"
fi

kexec -l -t bzImage "$kargs" "$reuse" --append="$*"

openrc-shutdown -K 0 # Replace with "systemd kexec" for Systemd init.
```

Although may require additional params from `/etc/default/grub` (or similar configuration used by your bootloader).

---

EDK II may also detect SD Card controller after soft boot so may setup an OS / bootloader on it as far as on the USB stick / disk.

MrChromebox, 2023-10-16:
there was a recent kernel change which caused the issue, reverting it resolves it. I've reached out to the patch author to try and figure out if it's something I can work around in firmware

MrChromebox, 2023-10-17:
and I'm working with Bayhub engineers to try and work around it from the firmware side

MrChromebox, 2023-10-18:
issue has been fixed, and will post updated images for Stoney boards once we've tested out the rest of the Windows enablement patches

MrChromebox, 2023-10-30:
fixed in MrChromebox-4.21.1 release

closed 2023-10-30 by MrChromebox